joan2937/pigpio #527: PWMdutycycle error message

asebian, 2022-04-14:
Hi,
I started using pigpio recently and I am very happy with it so far. Thanks for sharing it!

I am working on a stepper motor control application in C using PIGPIO_VERSION 79 shipped as package version 1.79-1+rpt1 with Raspian 11.3.

In order to test for an active PWM I call `gpioGetPWMdutycycle`.  If the return value is `PI_NOT_PWM_GPIO` (or `0`), then there is no PWM action going on at that pin, obviously. This works very well, but the library drops an error message at `stderr`. 
```
2022-04-14 23:48:32 gpioGetPWMdutycycle: not a PWM gpio (19)
```
It is really just a minor cosmetic issue. In my opinion, the user of the API should be allowed to ignore the error silently. But maybe there is a better way to do such a test I am not yet aware of.

Cheers!

guymcswain, 2022-04-20:
An obvious fix for your usage would be to change the switch default to
```c
default:
   return PI_NOT_PWM_GPIO;`
```
thus suppressing the error message.  However, that error message is useful for users running pigs communicating with the daemon as they get a human error message instead of -91. 
 I'm not convinced it is worthwhile but I'm open to other ideas.

```c
int gpioGetPWMdutycycle(unsigned gpio)
{
   unsigned pwm;

   DBG(DBG_USER, "gpio=%d", gpio);

   CHECK_INITED;

   if (gpio > PI_MAX_USER_GPIO)
      SOFT_ERROR(PI_BAD_USER_GPIO, "bad gpio (%d)", gpio);

   switch (gpioInfo[gpio].is)
   {
      case GPIO_PWM:
         return gpioInfo[gpio].width;

      case GPIO_HW_PWM:
         pwm = (PWMDef[gpio] >> 4) & 3;
         return hw_pwm_duty[pwm];

      case GPIO_HW_CLK:
         return PI_HW_PWM_RANGE/2;

      default:
         SOFT_ERROR(PI_NOT_PWM_GPIO, "not a PWM gpio (%d)", gpio);`
   }
}
```

asebian, 2022-04-20:
A message useful to application users ideally is generated at the application level contrary to library level. The decision to show an error-message to users of pigs should be made in pigs not the underlying API. Moving the error message to pigs could be a solution.

Currently the API does not seem to offer a method to query the PWM-status of a pin without potentially raising an error. Even turning PWM off is not all that obvious for people without knowledge of the internals. (Is it really off when dutycycle is zero? Isn't it even more off, when PWM is removed from that pin? Am I wasting resources?) 
Offering clear dedicated methods (in parallel) could be a solution.
```
int gpioPWMon(unsigned user_gpio, [unsigned dutycycle]); // deprecates gpioPWM
int gpioPWMoff(unsigned user_gpio); // short for gpioPWM(user_gpio, 0) maybe
int gpioPWMstatus(unsigned user_gpio); // like gpioGetPWMdutycycle but without errors thrown
```
What would I have to do to come up with a workaround? When my application starts up it does not know the state the pins are in. If it sets a pin to a defined state, is it guaranteed to stay in that state? Let us say I call `gpioPWM(pin,0)` at startup just to avoid the error message coming with `gpioGetPWMdutycycle(pin)`. Can the error reappear spontaneously (because the underlying [hardware]PWM got recycled and assigned to a different pin)?
Sure, I could keep track of the pin states. I could even to do shared memory IPC if I decided to use threads. But wouldn't that be more effort than fixing the API?

asebian, 2022-04-26:
I came across an issue with  the script engine but I am not convinced it is worthwhile reporting it.

guymcswain, 2022-05-14:
@asebian , I will look into suppressing the error message.  However, I looked into exposing the internal gpio info structure  but I don't believe it provides a complete status.  To make it 'complete' would require a lot of work and risk breaking stuff.

> When my application starts up it does not know the state the pins are in.

I see this as a general problem that the library can not solve for you.  When you invoke the library using `gpioInitialise`, the gpio states are unknown and therefore marked as 'undefined'.  You are permitted to invoke an API on any pin that is not masked.  The gpio status is not changed until you invoke an API that alters the gpio behavior.  So my assertion is that your application is responsible for the status of its gpio.

>  If it sets a pin to a defined state, is it guaranteed to stay in that state?

Yes, unless there is a bug or a client is connected and fiddling with the gpio.  pigpio does not support multi-user.