joan2937/pigpio #528: sigHandler: Unhandled signal 2, terminating

darshitagrawal, 2022-04-19:
I am working on interfacing dht11 with Raspberry Pi. My code is as per this link: https://github.com/jimstudt/dht11-pigpio/blob/master/dht11-pigpio.c. My concern is when I press Ctrl+C, I get the error "Unhandled Signal 2, terminating" and after that I cannot write further commands on my linux terminal. Can anyone tell me what should I do?

[image]

guymcswain, 2022-04-19:
By default, pigpio sets up to handle all signals but you have to attach a function to the signal - in this case SIGINT - using `gpioSetSignalFunc`.  Try `gpioSetSignal(SIGINT, cleanup)` and remove `atexit(cleanup)`.  (You may need to add exit() at the end of cleanup.)

Your other choice is the disable this default behavior and to handle the signal directly in your code.

darshitagrawal, 2022-04-27:
I tried putting this in my code. But I still cannot execute further commands in my linux terminal. I can type the commands plainly but they are not being executed. What should I do?

guymcswain, 2022-05-14:
@darshitagrawal , I have posted an improved signal handling topic branch on the repo named `signals`.  Can you test this issue against the `signals` branch?
```bash
wget https://github.com/joan2937/pigpio/archive/signals.zip
unzip signals.zip
cd pigpio-signals
make
sudo make install
```

closed 2023-04-28 by darshitagrawal