joan2937/pigpio #530: Running with a sample buffer size larger than the default (120) results in seg fault

lane-maxwell, 2022-04-28:
If I attempt to run the daemon with a sample buffer of anything larger than the default value, it results in a SIGSEGV unless I have DMA mode set to MBOX.  Below is the output from a gdb session.  I'm currently on master, commit `c33738a320a3e28824af7807edafda440952c05d`.  This behavior is occurring on the binary installed via apt (version 79) also.

```
# DMA mode set to the default (AUTO)
/usr/bin/pigpiod -b 200 -a 0 -g
2022-04-28 16:22:19 sigHandler: Unhandled signal 11, terminating
```

```
# DMA mode set to PMAP
/usr/bin/pigpiod -b 200 -a 1 -g
2022-04-28 16:23:31 sigHandler: Unhandled signal 11, terminating
```

```
# DMA mode set to MBOX - behaves properly
/usr/bin/pigpiod -b 200 -a 2 -g
```

```
cat /sys/firmware/devicetree/base/model 
Raspberry Pi 3 Model B Rev 1.2

 cat /proc/version
Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
```

```
/home/pi/pigpio# gdb pigpiod
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from pigpiod...
(gdb) run -s 2 -b 200 -f -g
Starting program: /home/pi/pigpio/pigpiod -s 2 -b 200 -f -g
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
initPCM (bits=10) at pigpio.c:7841
7841	   dmaIVirt[0]->periphData = 0x0F;
```

guymcswain, 2022-04-29:
This is 64-bit os.   Have you tried 32-bit?

lane-maxwell, 2022-04-29:
@guymcswain I just reimaged to a 32-bit OS and it works as expected, so this only exists in the 64-bit version.