joan2937/pigpio #533: Running without sudo

retostockli, 2022-05-26:
Is there any chance to make the pigpio c library (without the deamon) work without the sudo command?

I have moved all my code from wiringpi to pigpio, great  performance so far. Just the sudo really bugs me. I really would like to avoid running my user space code with the sudo command.

I know this has been asked before, so please close the request if you really can't do anything about it.

Sincerely
Reto

m-dhooge, 2022-06-02:
The library is like a kernel module: it directly reads and writes into memory to interact with the micro-controller of the Pi (https://github.com/joan2937/pigpio/issues/171#issuecomment-338127861 & https://github.com/joan2937/pigpio/issues/336)

If you don't want your app to have elevated privileges, either use `pigpiod` or [play with capabilities](https://stephanfr.blog/2021/05/02/managing-privileges-for-automated-raspberry-pi-gpio-testing/).

My 2 cents…

retostockli, 2022-06-03:
Thanks for the hint and the link to the other issues. I understand that accessing /dev/mem will always need root permission and /dev/gpiomem is not an option for pigpio. I guess that wiringPi used /dev/gpiomem then.

closed 2022-06-03 by retostockli