joan2937/pigpio #535: How to create a wave chain on two gpio simultaneously and with different frequencies

mahdigh782, 2022-05-31:
Hello
I try to run two stepper motor drivers at the same time.
Since synchronicity is important to me, I tried to run two wave chains at the same time, but when the second chain starts, the first chain stops.
Is there a way to solve my problem?
Or is there a library that produces the exact number of pulses and frequencies?
I also checked and found that the pwm software generated by the pigpiod is accurate but can not control the number of pulses. I also used the callback function but it does not respond at high frequencies.
Thank you

guymcswain, 2022-07-24:
> Is there a way to solve my problem?

Yes, very likely.  If you still have a need and you could not find a solution from searching past issues, then let me know and I will try to help you.

mahdigh782, 2022-08-15:
> Yes, very likely. If you still have a need and you could not find a solution from searching past issues, then let me know and I will try to help you.

Yes, I haven't set up the two stepper motor drives yet.
Just note that stepper motors operate at different frequencies and with different number of pulses.
Thanks

olewolf, 2025-02-22:
Reviving the question, I ran into the same issue today. I constructed two wave chains for two stepper motors connected to different GPIO pins. The two chains control the motors correctly if played one after the other. However, if I play one wave chain and start the second chain before the first one has completed, the first one is terminated--as the documentation of `wave_chain` appears to caution. I can make them run simultaneously with software PWM, but that's far too slow.

If possible, how can I run two waveform chains concurrently?