joan2937/pigpio #541: pigpio /dev/vcio not exist in 5.15.44

nicomedia, 2022-07-24:
Hello guys,

When i try to use pigpio in 5.15.44 kernel, i could not use pigpio because there is no /dev/vcio. When i try to modprobe vcio there is nothing change.

uname -a
Linux astroid 5.15.44-v7l+ #1 SMP Sun Jun 5 12:44:47 BST 2022 armv7l GNU/Linux

error : 
2022-07-23 12:33:11 initAllocDMAMem: mbox open failed(No such device or address)

when i try to upgrade system : 
sudo apt-get install pigpio
sudo apt-get update
sudo apt-get upgrade
sudo apt --fix-broken install
sudo reboot
sudo apt-get update
sudo apt-get upgrade

it returns to kernel 5.15.32 and pigpio is working correctly because /dev/vcio exists. 
But then i could not make my camera driver work imx327.

What could be the issue and how could i solve it  with kernel 5.15.44?

I know there is another issue related with this problem but the solution sudo modprobe vcio could not help me.

guymcswain, 2022-07-24:
> What could be the issue and how could i solve it with kernel 5.15.44?

Without /dev/vcio pigpio can't use the mailbox interface to mmap the peripherals required for functionality.  Not much we can do to help.

Piocky, 2022-12-05:
Hi @nicomedia, did you find a way to solve your issue? In my case, moving to a most recent kernel (from 5.4.72-v7l to 5.15.34-v7l), I also experimented the same issue on RPi CM4 rev 1.0, with following error when starting pigpiod:
`root@Piccolo-chambre:~# systemctl status pigpiod
Ã pigpiod.service - Daemon required to control GPIO pins via pigpio
     Loaded: loaded (/lib/systemd/system/pigpiod.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-12-02 20:14:20 CET; 24s ago
    Process: 12970 ExecStart=/usr/bin/pigpiod -l -x -1 -t 0 -s 1 (code=exited, status=0/SUCCESS)
   Main PID: 12971 (code=exited, status=1/FAILURE)

Dec 02 20:14:20 Piccolo-chambre systemd[1]: Starting Daemon required to control GPIO pins via pigpio...
Dec 02 20:14:20 Piccolo-chambre systemd[1]: Started Daemon required to control GPIO pins via pigpio.
Dec 02 20:14:20 Piccolo-chambre pigpiod[12971]: 2022-12-02 20:14:20 initAllocDMAMem: mbox open failed(No such device or address)
Dec 02 20:14:20 Piccolo-chambre pigpiod[12971]: Can't initialise pigpio library`

When adding -a1 parameter in pigpiod.service, everything's working well. But I recently received some RPi CM4 rev1.1, and with them I see some clock signals on my pins (40 kHz) when I switch them to output. If I remove -a1, then it works as expected. I really don't understand why I get clocks on those pins when switching to output, and also why with the same image, I've got /dev/vci0 on rev 1.1 and not 1.0

Piocky, 2022-12-05:
Okay, thanks for your answer anyway, I will try to reach people at Raspberry via their forum, as I think it's more related to the firmware than to pigpiod itself.

nicomedia, 2022-12-05:
Yes probably

closed 2022-12-05 by nicomedia