joan2937/pigpio #542: gpioSetISRFuncEx does not trigger on rpi3/4

folkertvanheusden, 2022-09-01:
Hi,

When making GPIO 2 high, the callback is **not** triggered on rpi4.

```
#include <pigpio.h>
#include <stdio.h>
#include <unistd.h>

void func(int gpio_n, int level, uint32_t tick, void *modemptr)
{
	printf("%d %d %d\n", gpio_n, level, tick);
}

int main(int argc, char *argv[])
{
	int gpio_n = 2;
	gpioInitialise();
    gpioSetMode(gpio_n, PI_INPUT);
    gpioSetISRFuncEx(gpio_n, RISING_EDGE, 10000, func, NULL);

    for(;;)
	    sleep(86400);

    return 0;
}
```

joan2937, 2022-09-05:
The ISR function uses the Linux sysfs system to provide interrupts.

I wonder if GPIO 2 has been already assigned for another purpose (e.g. I2C) and is not available to be used.

Check to see if **/dev/i2c-1** exists.

Also check that the return from gpioSetISRFuncEx is not an error code.

folkertvanheusden, 2022-09-08:
Hi,

No I2C configured and gpioSetISRFuncEx returned no error (0).

There's progress though: if I make sure that the device resets (makes low) its irq-pin and then set the isr-function, then it works.