joan2937/pigpio #545: Python distutils module is being discontinued

chrisl8, 2022-09-14:
According to https://peps.python.org/pep-0632/, and text output when running `make install`, the use of distutils is deprecated and distutils will soon be removed from Python.

distutils is used in `setup.py`

https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/setup.py#L3

Once distutils is gone `make install` will nolonger work as the `Makefile` calls `setup.py`
https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/Makefile#L107

bennuttall, 2022-09-14:
Yep, would recommend moving to setup.py or better, setup.cfg like so: https://github.com/piwheels/hostedpi/blob/main/setup.cfg

alphatech56, 2024-10-27:
It looks like the same problem when installing on OS Bookworm version on raspberry PI 

> _/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools._

_