joan2937/pigpio #548: Code Review

Frank-Friemel, 2022-10-16:
https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/pigpiod_if2.c#L345

Why is variable „got“ declared static? 

1 . Seems to be needless (to be static)
2. By declaring it static it‘s being shared by multiple callback-threads but I don‘t see a mutex to protect „got“

guymcswain, 2022-10-16:
Is there a specific issue you've encountered that lead to your code review?  I'd rather focus there first.

Frank-Friemel, 2022-10-16:
I've ported the client code to Windows and stumbled over this line. I guess that multiple remote servers won't be connected by one client, but if some user tries to do so ... I expect that it won't work

guymcswain, 2022-10-16:
> I guess that multiple remote servers won't be connected by one client (usually), but if some user tries to do so ... I expect that it won't work

It should.  If I get some time over the next month or so I may try a simple test case.  Would be great if you could generate a test case.

guymcswain, 2022-10-16:
Actually, there is at least one outstanding issue with multiple clients that is not related to the specific static variable you've identified.  The 'work around' for that issue, using the Python client, is to implement your threads behind the client.  So single client but supporting multi threads so to speak.

Frank-Friemel, 2022-10-16:
Well, you didn't answer my original question. Why is it declared "static"?

guymcswain, 2022-10-16:
I don't know.