joan2937/pigpio #551: [Question when switch from wirinPi to pigpio]: How to

hasenradball, 2022-11-24:
Hi,

I wanted to rework some libraries which working actually on wiringPi or other GPIO libs to get working with pigpio.
But I am not really familiar with where does the gpio support by the Libs come into account.
Can you give me some hints where I have to do the changes when switching to pigpio?

guymcswain, 2022-11-25:
You may receive better support on this question if you post it to the raspberrypi.org forums.

In short, my suggestion would be to write a set of C language wrapper functions that replace the wiring pi lib calls.  Inside these functions you would resolve any impedance mismatch between the two libraries:  parameter order and types, return values, etc.

That task will be time consuming and you may still need to restructure your original application code.  Pigpio offers a daemon that can accept calls from higher level scriptung languages such as Python and JavaScript.  It may be worth your time to explore rewriting your application if you are already familiar with one of the supported client languages.

guymcswain, 2023-01-20:
Question answered.

closed 2023-01-20 by guymcswain

hasenradball, 2023-01-20:
Hi, sorry for the late response.
A wrapper to use c library is no solution for me.

Tank you, i will aks this topic on the Pi forum.