joan2937/pigpio #553: DMA memory allocation mode change between CM4 rev 1.0 and 1.1

Piocky, 2022-12-01:
Hi,

I've got a problem that I don't know how to solve: I'm using a Raspberry Pi CM4 (so far I was using rev 1.0) on a project, that flash a microcontroller during boot if its software changed. For that I need to invoke the BSL of the µc, and use two pins for that purpose. Everything was working fine. I recently updated the kernel of my project from 5.4.72-v7l to 5.15.34-v7l, but when I did it, I also needed to add -a1 in the pigpiod parameters, otherwise I had the wollowing error:

> pigpiod[661]: 2022-12-01 17:27:17 initAllocDMAMem: mbox open failed(No such device or address)
> pigpiod[661]: Can't initialise pigpio library

But the thing is, I recently had some new CM4 rev 1.1, and if I keep this change, when I turn the two pins to pigpio.OUTPUT, i've got some weird signals on my pins, which contains 40kHz pulses. So obviously I can't invoke the BSL as wanted.

[image]

If I remove the -a1 it keeps working on rev 1.1, but then it doesn't work anymore 1.0, which is really anoying because I can't have two images in production, one for 1.0 and one 1.1

Does anybody aware of that problem, and (fingers crossed) know how to solve it?

guymcswain, 2022-12-01:
> I recently updated the kernel of my project from 5.4.72-v7l to 5.15.34-v7l, …

I’m interested in how you did the update.  The mail box interface should only be affected by the firmware associated with the Broadcom VPU.  I presume you only updated the Linux kernel.

Piocky, 2022-12-02:
> I’m interested in how you did the update. The mail box interface should only be affected by the firmware associated with the Broadcom VPU. I presume you only updated the Linux kernel.

I'm using Yocto project, and I moved from dunfell to kirkstone branch, so actually a lot of things changed. Do you know how can I check the mail box version?

guymcswain, 2023-01-20:
pigpio supports the Raspberry Pi OS(s).

closed 2023-01-20 by guymcswain