joan2937/pigpio #554: Problem pigpiod on Raspbian 11 (bulleye) with Relays PCB

pierol, 2022-12-29:
I'm using pigpio for interfacing with Wiegand reader and Relays PCB. I started this project on Raspberry PI 3 B+ installing Raspbian 10 (buster) and `pigpio` version 79 and it operated as expected. Yesterday i configured a new Raspberry PI 3 B+ with the last recommended version Raspbian 11 (bulleye). I then installed pigpio version 79 and started at boot time as service `pigpiod` and the relays switch on and off continusly and i must plug off the power to stop

guymcswain, 2022-12-30:
I have no idea of the problem but maybe more information may help.

What is the hardware revision?  64 or 32 bit OS?  Lite or desktop?

Can you run the pigpio tests successfully?

guymcswain, 2022-12-30:
FYI, to stop the service run
`sudo systemctl stop pigpiod`

pierol, 2022-12-30:
Thank you @guymcswain for your support. Here the information:
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 000000000e07251a
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
```
The OS is Raspbian 11 (bulleye) 32bit Desktop.
I've never run the pigpio test how can i do that ?
At the moment i've reinstalled the Raspbian 10 (buster) and all is operated as expected.

pierol, 2022-12-31:
I've made some test as reported in the pigpio site.

OS release 32bit:

```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
Hardware Revision:

```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 00000000a249a97c
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
```

This is `sudo ./x_pigpio # check C I/F`

```
Testing pigpio C I/F
[... 622 lines cut ...]
WRITE-c ok
WRITE-d ok
WVCLR ok
WVAS ok
WVAG ok
WVCRE ok
WVTX ok
WVBSY-a ok
WVBSY-b ok
WVHLT ok
WVBSY-c ok
WVTXR ok
WVBSY-d ok
WVHLT ok
WVBSY-e ok
WVSC-a ok
WVSC-b ok
WVSC-c ok
WVSM-a ok
WVSM-b ok
WVSM-c ok
WVSP-a ok
WVSP-b ok
WVSP-c ok
```

guymcswain, 2022-12-31:
Thank you for the test results!

So, it appears that waveforms are broken in this bullseye release.  If you could help nail this down further with:
`cat /etc/debian_version`
and 
` uname -a`

Waveforms is essentially the same as pwm (DMA based control).  Can we confirm that your application is using a DMA based control API?  That would be nice to know there is consistency between the test failures and your application behavior.

pierol, 2022-12-31:
Hi @guymcswain thank you for your support. Here the results.
` cat /etc/debian_version`:
11.5
`uname -a`:
`Linux UC19Domotica 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux

To test the behavior:

1. I've enable the pigpiod to start at boot time.
2. Then in `/etc/rc.local` i put this init code for the two GPIO i used:

`

guymcswain, 2022-12-31:
From the above code snippet, it looks like you are only using gpio read and write APIs.  Those APIs have nothing to do with the DMA.  The basic tests above indicate basic gpio read write are working - Test suite 1.x - so this is inconsistent with your application failures. 

It would be good to know if GPIO is using glitch or noise filtering which does rely on DMA samples and would presumably fail since Test suite 13.x shows failures in this area.

Another thing just caught my eye.  You say the OS is Raspbian 11 and not Raspberry Pi OS.  If that is correct, can you test with the official RPi OS.

pierol, 2022-12-31:
I created the SD card for raspberry using the official way with the Raspberry Pi Imager and I’ve choose the recommended 32 bit OS image so is the Raspberry Pi Official OS. The problem come out just booting with the configuration i explained without executing my codeInviato da iPhone

guymcswain, 2022-12-31:
My misunderstanding, I thought they were separate variants.  I would prefer to change the title to RPi OS 11 in this case.

pierol, 2022-12-31:
I will change the title as you suggestedInviato da iPhone

guymcswain, 2023-01-01:
In case of a conflict with audio please try the following:

- Ensure you have remote access (ssh).
- Unplug your HDMI cable from the rpi.
- Re-run the pigpio tests over a remote connection.

Background:  Audio may use PWM hardware registers which conflict with pigpio.  RPi OS 11 release notes mention: "Modifications to HDMI audio output selection to support the above".

pierol, 2023-01-01:
Thank you @guymcswain for your suggestions.
The tests i've reported above has been done in that condiftions:

- via remote access SSH
- HDMI cable unplugged

guymcswain, 2023-01-19:
@pierol , thank you for your patience.

I don't have my hardware freed up yet to do any testing.  However, I recently did a search on the raspberrypi.com forums and found an issue that may be related to yours.  [Here](https://forums.raspberrypi.com/viewtopic.php?p=2067847&hilit=pigpio#p2067847) is a link to that thread.

It would be interesting to see if your problem goes away when changing to the Lite OS version of Bulleye.  Seeing that you are running your testing remotely perhaps this could be a long term workaround for your situation.

pierol, 2023-01-19:
@guymcswain Thank you for your support. Yes it seems to be the same issue.
I've done the same config (RPi OS 11 32 bit Desktop) with a Raspberry Pi 4B 2G and all operate as expected without the problem.
In the forum one says it has the problem even on rpi4 but he has installed a RPi OS 11 64bit Desktop.

guymcswain, 2023-01-19:
Please try the following:
```
// Create a drop-in file for pigpiod.service:
// note: automatically creates '/etc/systemd/system/pigpiod.service.d/override.conf'
# sudo systemctl edit pigpiod

// Add this content to the file:
[Install]
WantedBy=
WantedBy=default.target
```
After this it should restart the daemon or it will prompt you to do so.

pierol, 2023-01-20:
Sorry @guymcswain I should add this 
```
[Install]
WantedBy=
WantedBy=default.target
```
or this:
```
[Install]
WantedBy=default.target
```

guymcswain, 2023-01-20:
You need both statements.  The first overrides the existing setting then
the second sets the new setting.

pierol, 2023-01-20:
> @pierol , thank you for your patience.
>
> I don't have my hardware freed up yet to do any testing. However, I recently did a search on the raspberrypi.com forums and found an issue that may be related [...]

Yes i can confirm that with RPi OS 11 Lite bullseye the problem disappears

pierol, 2023-01-20:
> You need both statements. The first overrides the existing setting then the second sets the new setting.
> […](#)

I tried with this:
```
[Install]
WantedBy=
WantedBy=default.target

```
but the problem persists

guymcswain, 2023-01-22:
I'm got my pi3b+ (a020d3) with a fresh install of RPi OS 32 bit desktop: 
```
cat /etc/debian_version
11.6
```
I tried to reproduce the experiments on the raspberrypi forum, using an LED on gpio 17, but so far I'm not able to reproduce the flashing behavior.  However, I have experienced the system to hang on reboot occasionally.  The screen goes blank, and I cannot do any debug at that point.

Do you have any suggestions on how to reproduce the erratic gpio behavior?  Which gpio pins are used in your script?  

Meanwhile, I'm researching how to make the daemon start later in the boot process.

guymcswain, 2023-01-22:
Ok, I think I understand how to get the pigpiod service to start later in the boot sequence.  But first, I should explain why:  My theory is that pigpiod service is starting too soon and possibly conflicting with another process that is touching peripheral registers that pigpio also uses.

1.  Since we previously changed the pigpiod.service on your system, we need to revert back to the original by doing:  `sudo systemctl revert pigpiod`
2. List the service file using `systemctl cat pigpiod` and it should look like this:
```
# /lib/systemd/system/pigpiod.service
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```
3. Create a drop-in override file to modify the pigpiod service.  `sudo systemctl edit pigpiod`
4. Add these lines:
```
[Unit]
Requires=default.target
After=default.target
```
5. `sudo systemctl daemon-reload`
6. Verify the service is still enabled by `sudo systemctl status pigpiod`
7. If it is not, then enable it by `sudo systemctl enable pigpiod`
8. 'sudo reboot'

Please let me know if your script is now functioning properly.  Also, verify that pigpiod.service is running near the end of the boot sequence.  To do so, run `systemd-analyze blame` (you may need to `apt install systemd-analyze`).  In the output from this command you should see 'pigpiod.service' listed near the end of the boot sequence.  On my system it shows up after 'nfs-config.service'.

pierol, 2023-01-22:
After done the steps you suggested the deamon doesn't start at boot time. Here le output of some commands:

1. `systemctl cat pigpiod`:
```
# /lib/systemd/system/pigpiod.service
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target

# /etc/systemd/system/pigpiod.service.d/override.conf
[Unit]
Requires=default.target
After=default.target
```

2.  `sudo systemctl status pigpiod`:
```
● pigpiod.service - Daemon required to control GPIO pins via pigpio
     Loaded: loaded (/lib/systemd/system/pigpiod.service; enabled; vendor prese>
    Drop-In: /etc/systemd/system/pigpiod.service.d
             └─override.conf
     Active: inactive (dead)
```

3.  If i make a `ps -aux | grep pigpiod` the proccess doesn't appear.
4. ` systemd-analyze blame` the pigpiod doesn't appear
```
9.191s hciuart.service
3.930s rc-local.service
2.534s dev-mmcblk0p2.device
2.467s fstrim.service
2.082s raspi-config.service
1.678s udisks2.service
1.578s ModemManager.service
1.286s cups.service
1.183s ssh.service
1.133s plymouth-read-write.service
1.118s glamor-test.service
1.045s avahi-daemon.service
 998ms rng-tools-debian.service
 891ms keyboard-setup.service
 842ms networking.service
 815ms systemd-timesyncd.service
 793ms polkit.service
 786ms systemd-logind.service
 735ms dphys-swapfile.service
 697ms e2scrub_reap.service
 686ms gldriver-test.service
 656ms lightdm.service
 649ms user@1000.service
 608ms systemd-udev-trigger.service
 490ms dhcpcd.service
 485ms systemd-journal-flush.service
 472ms systemd-udevd.service
 454ms systemd-fsck@dev-disk-by\x2dpartuuid-5c76fb9e\x2d01.service
 446ms systemd-fsck-root.service
 445ms systemd-journald.service
 403ms wpa_supplicant.service
 346ms systemd-tmpfiles-setup-dev.service
 345ms packagekit.service
 343ms modprobe@drm.service
 310ms rsyslog.service
 237ms rpi-eeprom-update.service
 234ms modprobe@fuse.service
 213ms bluetooth.service
 208ms systemd-remount-fs.service
 201ms systemd-modules-load.service
 200ms systemd-tmpfiles-setup.service
 186ms fake-hwclock.service
 185ms systemd-update-utmp.service
 162ms dev-mqueue.mount
 157ms run-rpc_pipefs.mount
 154ms modprobe@configfs.service
 151ms sys-kernel-debug.mount
 145ms sys-kernel-tracing.mount
 133ms systemd-user-sessions.service
 124ms kmod-static-nodes.service
 121ms systemd-random-seed.service
 103ms systemd-sysusers.service
```

guymcswain, 2023-01-22:
The service is dead.  Make sure you reloaded the systemd daemon - step 4.  Then do step 7 for good measure.

pierol, 2023-01-22:
I did step 4. and step 5.

guymcswain, 2023-01-22:
Something strange between your system and mine:  On my system 'rc-local.service' runs late in the sequence.  On your system it the second in sequence!  Or maybe I don't understand the sequence - maybe its in reverse time?  But it seems that rc-local.service should run near the end of the boot sequence.

guymcswain, 2023-01-22:
I can only suggest to power down then check the listing and the status on pigpiod.service.

pierol, 2023-01-22:
It seems that making the override break somthing and the deamon doesn't start anymore. If i make a `sudo systemctl revert pigpiod` and after `sudo systemctl daemon-reload` and `sudo systemctl enable pigpiod` the daemon started and the problem appears.

guymcswain, 2023-01-22:
You can edit the complete file skipping the override by doing `sudo systemctl edit pigpiod --full`.
Then when you `systemctl cat pigpiod` it will only show the service file under /etc/systemd/system (not /lib/systemd/system).  

/etc/systemd/system/pigpiod.service should look like this:

```
[Unit]
Description=Daemon required to control GPIO pins via pigpio
Requires=default.target
After=default.target
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```
Or just create a text file name pigpiod.service and copy it into /etc/systemd/system directly yourself.

And finally, `sudo systemctl daemon-reload` is important.

pierol, 2023-01-22:
There is something wrong with these two line in the file `pigpiod.service` inside `/etc/systemd/system`:
```
Requires=default.target
After=default.target
```
When i remove them the deamon started when i insert them the daemon doesn't started at boot time.
Making the procedure, after modify the file, of

1. `sudo systemctl daemon-reload`
2.  `sudo systemctl enable pigpiod`

pierol, 2023-01-22:
> I'm got my pi3b+ (a020d3) with a fresh install of RPi OS 32 bit desktop:
>
> ```
> cat /etc/debian_version
> 11.6
> ```
>
> I tried to reproduce the experiments on the raspberrypi forum, using an LED on gpio [...]

To reproduce the problem, after enabled the pigpiod deamon at boot time, try to set a GPIO (in my case GPIO 27) even after the boot. When you try to se the GPIO the operate green led of the raspberry pi flashes

guymcswain, 2023-01-22:
Does `journalctl -u pigpiod` show any problem?

pierol, 2023-01-22:
No, this is when started without that two mentionted lines
```
-- Boot 8e59563b9f0a49508d75998b0f96cefe --
Jan 22 21:25:06 raspberrypi systemd[1]: Starting Daemon required to control GPIO pins via pigpio...
Jan 22 21:25:06 raspberrypi systemd[1]: Started Daemon required to control GPIO pins via pigpio.
```

pierol, 2023-01-23:
Answer to your question:

- Q1 ? All the tests i've done and i'm doing are on a fresh install without running any of my serivices. All i do is to setup in `/etc/rc.local` the GPIO 27 and GPIO 22 where i run a python3 script that do this initialization:
```bash
     if [ "$_IP" ]; then
       printf "My IP address is %s\n" "$_IP"
     fi
     export PYTHONPATH=/home/pi/sanpietro
     /usr/bin/python3 /home/pi/sanpietro/sanpietroInit.py
     
     exit 0

   ```
the `sanpietroInit.py` do this:
```python
 import RPi.GPIO as GPIO
   if __name__ == '__main__':
       # Set each pin as an output and make it low:
       GPIO.setmode(GPIO.BCM)
       GPIO.setup(27, GPIO.OUT)
       GPIO.output(27, GPIO.HIGH)
       GPIO.setup(22, GPIO.OUT)
       GPIO.output(22, GPIO.HIGH)
```
- Q2 ? the output of the command `systemd-analyze critical-chain`
```
graphical.target @12.039s
└─multi-user.target @12.039s
  └─getty.target @12.038s
    └─getty@tty1.service @12.037s
      └─plymouth-quit-wait.service @12.025s +6ms
        └─rc-local.service @7.901s +4.118s
          └─network.target @7.843s
            └─wpa_supplicant.service @7.311s +525ms
              └─dbus.service @6.479s
                └─basic.target @6.448s
                  └─sockets.target @6.446s
                    └─triggerhappy.socket @6.445s
                      └─sysinit.target @6.410s
                        └─systemd-timesyncd.service @5.791s +617ms
                          └─systemd-tmpfiles-setup.service @5.561s +212ms
                            └─local-fs.target @5.501s
                              └─run-user-1000-gvfs.mount @15.908s
                                └─run-user-1000.mount @11.916s
                                  └─local-fs-pre.target @2.726s
                                    └─systemd-tmpfiles-setup-dev.service @2.617s +107ms
                                      └─systemd-sysusers.service @2.532s +79ms
                                        └─systemd-remount-fs.service @2.307s +200ms
                                          └─systemd-fsck-root.service @1.826s +471ms
                                            └─systemd-journald.socket @1.718s
                                              └─-.mount @892ms
                                                └─-.slice @892ms
```
-Q3 the output of the command `systemd-analyze critical-chain pigpiod.service`
```terminal
pigpiod.service +720ms
└─basic.target @6.448s
  └─sockets.target @6.446s
    └─triggerhappy.socket @6.445s
      └─sysinit.target @6.410s
        └─systemd-timesyncd.service @5.791s +617ms
          └─systemd-tmpfiles-setup.service @5.561s +212ms
            └─local-fs.target @5.501s
              └─run-user-1000-gvfs.mount @15.908s
                └─run-user-1000.mount @11.916s
                  └─local-fs-pre.target @2.726s
                    └─systemd-tmpfiles-setup-dev.service @2.617s +107ms
                      └─systemd-sysusers.service @2.532s +79ms
                        └─systemd-remount-fs.service @2.307s +200ms
                          └─systemd-fsck-root.service @1.826s +471ms
                            └─systemd-journald.socket @1.718s
                              └─-.mount @892ms
                                └─-.slice @892ms
```

guymcswain, 2023-01-23:
I think your rc-local.service has a dependency on pigpiod (via import RPi.GPIO) and is waiting for the socket to open.  But when we make pigpiod wait until after default.target (equivalent to graphical.target on desktop) the socket will never open.

guymcswain, 2023-01-23:
rc-local.service runs before the graphical.target.  Your script needs to run after graphical.target.  I've read on the forums that autostart may be what you need to use.  But I would highly recommend just use a systemd service for your script.

pierol, 2023-01-23:
I tried this:

I made this:
- I've deleted the two lines in `/etc/rc.local` so that i do nothing during boot time.
- The problem appears anyway: there is only pigpiod started at boot time.

The only way to stop flashing is to disable the pigpiod.service.

guymcswain, 2023-01-23:
You said you could not boot with the modified pigpiod.service I proposed.  So now you are saying it boots but doesn't solve the "flashing" issue (after you have disabled your lines in /etc/rc.local)?

guymcswain, 2023-01-23:
You can create a systemd service file similar to this:  (it may need a dependency on pigpiod)
```
[Unit]
Description=Your Python script
After=default.target

[Service]
ExecStart=/usr/bin/python3 /home/pi/sanpietro/sanpietroInit.py

[Install]
WantedBy=multi-user.target
```

pierol, 2023-01-23:
With the modified pigpiod.service you proposed the service doesn't start. So i'm still with the original pigpiod.service file.

> So now you are saying it boots but doesn't solve the "flashing" issue (after you have disabled your lines in /etc/rc.local)?

Yes, it boots with the flashing problem but not all the times: on 4 boots 2 didn't flash, 2 did flashes

pierol, 2023-01-23:
If you want i can permit you to make a AnyDesk session on my desktop to see why it doesn't start

guymcswain, 2023-01-23:
No thank you.  But I will help you write a systemd service file.

You need to cleanly re-install RPi OS bullseye without adding services.  Then make the changes to pigpiod service file.  Then tell me if your system is unstable.

pierol, 2023-01-23:
On question, when you install the RPi OS Desktop bullseye 32 bit you made the updates ?

pierol, 2023-01-23:
I find in the system log this:

 `multi-user.target: Job pigpiod.service/start deleted to break ordering cycle starting with multi-user.target/start`

guymcswain, 2023-01-23:
I did a clean install then tried to reproduce the gpio 'flashing' as described on the rpi forum thread.  After I couldn't reproduce it, I upgraded the OS with all the latest.  After that I started to see some instability.  But I can't say for certain it was due to the upgrade.  In either case, after I changed the pigpiod.service I've had stability.

pierol, 2023-01-23:
Yes, is the version i'm using now but the service is deleted during the boot time for a cycle dependency.

guymcswain, 2023-01-23:
Is this the new install?

pierol, 2023-01-23:
> Is this the new install?

No, at the moment i've no time to do that. It's the installation i've made yesterday.

guymcswain, 2023-01-23:
```
sudo systemctl daemon-reload
sudo reboot

#after boot-up
systemctl cat pigpiod
systemd-analyze critical-chain pigpiod.service
```

pierol, 2023-01-23:
`systemctl cat pigpiod`
```
[Unit]
Description=Daemon required to control GPIO pins via pigpio
Requires=default.target
After=default.target
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```
`systemd-analyze critical-chain pigpiod.service`
```
The time when unit became active or started is printed after the "@" character.
The time the unit took to start is printed after the "+" character.

└─graphical.target @8.991s
  └─multi-user.target @8.991s
    └─ModemManager.service @7.758s +1.230s
      └─polkit.service @6.642s +1.103s
        └─basic.target @6.368s
          └─sockets.target @6.368s
            └─triggerhappy.socket @6.367s
              └─sysinit.target @6.317s
                └─systemd-timesyncd.service @5.665s +650ms
                  └─systemd-tmpfiles-setup.service @5.384s +267ms
                    └─local-fs.target @5.312s
                      └─run-user-1000.mount @11.606s
                        └─local-fs-pre.target @2.762s
                          └─systemd-tmpfiles-setup-dev.service @2.657s +103ms
                            └─systemd-sysusers.service @2.549s +102ms
                              └─systemd-remount-fs.service @2.335s +189ms
                                └─systemd-fsck-root.service @1.836s +492ms
                                  └─systemd-journald.socket @1.725s
                                    └─system.slice @938ms
                                      └─-.slice @938ms
```

guymcswain, 2023-01-23:
pigpiod.service is not listed?
do `systemctl status pigpiod`
then `journalctl -u pigpiod --no-pager | tail -20`

pierol, 2023-01-23:
```
 pigpiod.service - Daemon required to control GPIO pins via pigpio
     Loaded: loaded (/etc/systemd/system/pigpiod.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
```

```
-- Boot f3c8f5ea0e274e8780992cb04e292c61 --
Jan 23 20:41:58 raspberrypi systemd[1]: Starting Daemon required to control GPIO pins via pigpio...
Jan 23 20:41:59 raspberrypi systemd[1]: Started Daemon required to control GPIO pins via pigpio.
Jan 23 20:55:22 raspberrypi systemd[1]: multi-user.target: Found ordering cycle on graphical.target/stop
Jan 23 20:55:22 raspberrypi systemd[1]: multi-user.target: Found dependency on pigpiod.service/stop
Jan 23 20:55:22 raspberrypi systemd[1]: multi-user.target: Found dependency on multi-user.target/stop
Jan 23 20:55:22 raspberrypi systemd[1]: multi-user.target: Job graphical.target/stop deleted to break ordering cycle starting with multi-user.target/stop
Jan 23 20:55:22 raspberrypi systemd[1]: Stopping Daemon required to control GPIO pins via pigpio...
Jan 23 20:55:23 raspberrypi systemd[1]: pigpiod.service: Sent signal SIGTERM to control process 1334 (systemctl) on client request.
Jan 23 20:55:23 raspberrypi systemd[1]: pigpiod.service: Sent signal SIGTERM to main process 394 (pigpiod) on client request.
Jan 23 20:55:23 raspberrypi systemd[1]: pigpiod.service: Control process exited, code=killed, status=15/TERM
Jan 23 20:55:23 raspberrypi systemd[1]: pigpiod.service: Main process exited, code=exited, status=255/EXCEPTION
Jan 23 20:55:23 raspberrypi systemd[1]: pigpiod.service: Failed with result 'signal'.
Jan 23 20:55:23 raspberrypi systemd[1]: Stopped Daemon required to control GPIO pins via pigpio.
Jan 23 20:55:23 raspberrypi systemd[1]: pigpiod.service: Consumed 51.560s CPU time.
```

guymcswain, 2023-01-23:
This makes no sense to me.  Something else is conflicting but I don't know how to debug it.  I will wait until after you have a clean install.

guymcswain, 2023-01-23:
I have been doing all of my testing from a local console if that makes any difference.

guymcswain, 2023-01-23:
I'm plugged in to a monitor with gui but only using terminal.  It may be worth trying locally just to see if there is a difference,

pierol, 2023-01-23:
>  It may be worth trying locally just to see if there is a difference,

No any difference the service doens't start

pierol, 2023-01-25:
Hi @guymcswain,
i've a new installation with updates applied. 
Tell me the steps i should do to configure the pigpiod deamon. At the moment i've done nothing neither `sudo systemctl enable pigpiod`. As you know the pigpiod is already installed in this version of the RPi OS 32 bit Desktop bullseye.
The only difference respect to your board is the you installed the udates after the installazion of the OS

guymcswain, 2023-01-25:
Ok, let's first confirm somethings:
`pigpoid -v` should return "79"
`pigs pigpv` should return "socket not connected"
`systemctl status pigpiod` should show that pigpiod is not enabled
`systemctl cat pigpiod` shows only the unit file under /lib/systemd/system

pierol, 2023-01-25:
Can i do all these steps via SSH or locally ?

guymcswain, 2023-01-25:
ssh is ok

pierol, 2023-01-25:
> Ok, let's first confirm somethings: 
> `pigpoid -v` should return "79" 
confirmed
> `pigs pigpv` should return "socket not connected" 
this return "socket connect failed"

> `systemctl status pigpiod` should show that pigpiod is not enabled 
```
● pigpiod.service - Daemon required to control GPIO pins via pigpio
     Loaded: loaded (/lib/systemd/system/pigpiod.service; disabled; vendor pres>
     Active: inactive (dead)
```

>`systemctl cat pigpiod` shows only the unit file under /lib/systemd/system

```
# /lib/systemd/system/pigpiod.service
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target

```

[33 more replies]