joan2937/pigpio #555: Expose the board serial number

kurisuD, 2023-01-03:
Hello Joan,

In order to ease development of pigpio dependent Home Assistant integrations, specifically regarding [their guidelines on entities unique identifiers](https://developers.home-assistant.io/docs/entity_registry_index#example-acceptable-sources-for-a-unique-id), I would like to suggest exposing the board serial number.

Would you consider a pull request ?

guymcswain, 2023-01-20:
If it easy for me to understand, is of very limited (small) code changes and passes the suite of tests, then yes, I will consider it.

kurisuD, 2023-01-23:
Hello Guy

Thanks a lot for your feedback. I believe all the conditions are met, so went ahead with #558