joan2937/pigpio #556: RPI4 gpioTerminate() Hangs

wschopohl, 2023-01-19:
[guymcswain](https://github.com/guymcswain) asked me to create a new issue for this - before I was adding comments to issue [#478](https://github.com/joan2937/pigpio/issues/478)

I'm using pigpio library on a raspberry 4 with a buildroot aarch64 custom image using BusyBox as init system.

**Problem:**
I had problems with a simple test program, that would work fine when running, but would hang once I called `gpioTerminate()`.

**Inspection:**
When trying to pinpoint the error, I started with the pigpiod application. The pigpiod application would also hang when I sent it a SIGHUP or SIGTERM signal - although reading the source code it should terminate nicely also using the `gpioTerminate()` call.

After investigating a bit more with my limited gdb skills, it seems like the fifo interface is causing problems in my case. When I start pigpiod with the -f option, I can stop it with a SIGTERM signal.

**Solution:**
That means in my program I can use this code to disable the fifo interface and everything works as expected:

`gpioCfgInterfaces(PI_DISABLE_FIFO_IF); // call before gpioInitialise()`

Calling `gpioTerminate()` will not hang anymore and I can stop my test program nicely.

**Out Words:**
In my case this solution is sufficient, because I don't need a fifo interface. I opened this issue to make you guys aware of this and to offer a solution to users that might end up with a similar problem.

kiedtl, 2024-12-22:
I have this exact issue on a Raspberry Pi Zero running Alpine. I'd assume this occurs on all Pi's then, unfortunately.

guymcswain, 2024-12-22:
You should try the ‘develop’ branch; It has an improved signal handler.

guymcswain, 2024-12-23:
I'm glad it resolved your issue.  I probably will.  I was waiting for some feedback and then sort of lost interest for a while.