joan2937/pigpio #559: bind to port failed (Cannot assign requested address)

nigeljohnson73, 2023-01-29:
I read item #298 and it says this issue is fixed in version 79.

I am running a Raspberry pi 4b 2GB with Raspian lOS lite, and I still get the issue with the daemon not starting. I've checked but nothing is running on port 8888 already, and I manually downloaded and installed the latest version (79), several reboots and manual attempts at starting the service and running it on the command line and this is what I get (the simplest script to reproduce)

    pi@asciipi:~ $ /usr/bin/pigpiod -v
    79
    pi@asciipi:~ $ sudo /usr/bin/pigpiod -l
    pi@asciipi:~ $ 2023-01-29 12:36:42 initInitialise: bind to port 8888 failed (Cannot assign requested address)
    Can't initialise pigpio library
    pi@asciipi:~ $ netstat | grep 8888
    pi@asciipi:~ $ 

What am I missing or what have I not read?

guymcswain, 2023-01-29:
Should work.  Can you provide details on the OS, how you installed it.  Hardware revision too.

guymcswain, 2023-01-29:
Have you tried leaving off `-l`

nigeljohnson73, 2023-01-29:
I was originally excited there wasa lock file... but removing it didn't help :(

> pi@asciipi:~ $ sudo /usr/bin/pigpiod
> 2023-01-29 19:55:14 initInitialise: Can't lock /var/run/pigpio.pid
> Can't initialise pigpio library
> pi@asciipi:~ $ ls -als  /var/run/pigpio.pid
> 4 -rw-r--r-- 1 root root 5 Jan 29 19:55 /var/run/pigpio.pid
> pi@asciipi:~ $ sudo rm /var/run/pigpio.pid
> pi@asciipi:~ $  sudo /usr/bin/pigpiod
> pi@asciipi:~ $ 2023-01-29 19:56:19 initInitialise: bind to port 8888 failed (Address already in use)
> Can't initialise pigpio library
> 
> pi@asciipi:~ $ !ls
> ls -als  /var/run/pigpio.pid
> ls: cannot access '/var/run/pigpio.pid': No such file or directory
> pi@asciipi:~ $

nigeljohnson73, 2023-01-29:
OS is the latest Raspian lite installed through the Raspberry PI imager. This is the general first login path I take:

> sudo apt update -y
> sudo apt full-upgrade -y
> sudo apt autoremove -y
> sudo rpi-eeprom-update -a -d
> cat >/tmp/boot.conf <<EOF
> [all]
> BOOT_UART=0
> WAKE_ON_GPIO=1
> ENABLE_SELF_UPDATE=1
> BOOT_ORDER=$BOOT
> EOF
> sudo rpi-eeprom-config --apply /tmp/boot.conf
> sudo apt install -y lsb-release apt-transport-https ca-certificates git python3-dev python3-pip python3-pil automake autoconf pkg-config libcurl4-openssl-dev libjansson-dev libssl-dev libgmp-dev make g++ tor screen xserver-xorg x11-xserver-utils xinit openbox chromium-browser xserver-xorg-input-evdev libhdf5-dev libhdf5-serial-dev python3-pyqt5 libatlas-base-dev libjasper-dev libgtk2.0-dev pkg-config samba samba-common-bin cups build-essential libcups2-dev libcupsimage2-dev python3-serial pigpio python3-pigpio
> sudo pip install --upgrade pip setuptools wheel opencv-python qrcode[pil] pigpio

I have also installed the zj-58 printer drivers for a thermal printer I have attached

guymcswain, 2023-01-29:
I have not played with `rpi-eeprom-update`.   Maybe something new  there.

So you start with lite OS but set it up as a desktop?  Are you using Systemd?  If a fresh install, pigpiod should be disabled.

nigeljohnson73, 2023-01-29:
There is no desktop, I launch chromium browser though. I do not actively use systemd, but it may be there and doing it's thing. I believe I used it to enable the service at boot as per the instructions.

guymcswain, 2023-01-29:
Which service, pigpiod?

nigeljohnson73, 2023-01-30:
The [instructions I followed](https://gist.github.com/tstellanova/8b1fb350a148eace6541b5fbd2c021ca) said to enable the daemon on startup, so I did that. It fails.

When I check the status, it's failed, and then all the rest above

```
pi@asciipi:~ $ sudo service pigpiod status
● pigpiod.service - Daemon required to control GPIO pins via pigpio
     Loaded: loaded (/lib/systemd/system/pigpiod.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2023-01-29 12:35:53 GMT; 18h ago
    Process: 1064 ExecStart=/usr/bin/pigpiod -l (code=exited, status=0/SUCCESS)
   Main PID: 1065 (code=exited, status=1/FAILURE)
        CPU: 62ms

Jan 29 12:35:53 asciipi systemd[1]: Starting Daemon required to control GPIO pins via pigpio...
Jan 29 12:35:53 asciipi systemd[1]: Started Daemon required to control GPIO pins via pigpio.
Jan 29 12:35:53 asciipi pigpiod[1065]: 2023-01-29 12:35:53 initInitialise: bind to port 8888 failed (Cannot assign requested address)
Jan 29 12:35:53 asciipi pigpiod[1065]: Can't initialise pigpio library
Jan 29 12:35:53 asciipi systemd[1]: pigpiod.service: Main process exited, code=exited, status=1/FAILURE
Jan 29 12:35:53 asciipi systemd[1]: pigpiod.service: Failed with result 'exit-code'.
pi@asciipi:~ $ 
```

Yep, pigs returns the current version

```
pi@asciipi:~ $ pigs pigpv
79
pi@asciipi:~ $ 
```

guymcswain, 2023-02-01:
Try disabling the service, reboot, then run the daemon from command line.  The fact that pigs returned the version indicates the service was already running, in which case trying to run the daemon again will fail.

nigeljohnson73, 2023-02-02:
So I have done another few clean installs of Raspian OS lite 32 bit, update, upgrade etc, and I still got a failure of the service. and my code still throws an exception and the service won't start. I was about to give up and then I realised what you actually said, that if the version was returned it was working. So I tried again on the new install and it fails again.

```
pi@asciipi:/webroot/asciipi $ pigs pigpv
socket connect failed
pi@asciipi:/webroot/asciipi $ pigs HWVER
socket connect failed
```

Due to it being port thing, I thought I'd give a change of port a go...

```
pi@asciipi:/webroot/asciipi $ sudo /usr/bin/pigpiod -l -p 8887
pi@asciipi:/webroot/asciipi $ 2023-01-31 07:09:05 initInitialise: bind to port 8887 failed (Cannot assign requested address)
Can't initialise pigpio library

pi@asciipi:/webroot/asciipi $ sudo /usr/bin/pigpiod -l -p 8886
pi@asciipi:/webroot/asciipi $ 2023-01-31 07:09:11 initInitialise: bind to port 8886 failed (Cannot assign requested address)
Can't initialise pigpio library

pi@asciipi:/webroot/asciipi $ sudo /usr/bin/pigpiod -l -p 8885
pi@asciipi:/webroot/asciipi $ 2023-01-31 07:09:16 initInitialise: bind to port 8885 failed (Cannot assign requested address)
Can't initialise pigpio library

pi@asciipi:/webroot/asciipi $ 
```

The reason I'm using the -l is because the EXECSTART in the service definition says it uses it

```
pi@asciipi:/webroot/asciipi $ sudo service pigpiod status
● pigpiod.service - Daemon required to control GPIO pins via pigpio
     Loaded: loaded (/lib/systemd/system/pigpiod.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-01-31 06:58:55 GMT; 14min ago
    Process: 2136 ExecStart=/usr/bin/pigpiod -l (code=exited, status=0/SUCCESS)
   Main PID: 2137 (code=exited, status=1/FAILURE)
        CPU: 71ms

Jan 31 06:58:55 asciipi systemd[1]: Starting Daemon required to control GPIO pins via pigpio...
Jan 31 06:58:55 asciipi systemd[1]: Started Daemon required to control GPIO pins via pigpio.
Jan 31 06:58:55 asciipi pigpiod[2137]: 2023-01-31 06:58:55 initInitialise: bind to port 8888 failed (Cannot assign requested address)
Jan 31 06:58:55 asciipi pigpiod[2137]: Can't initialise pigpio library
Jan 31 06:58:55 asciipi systemd[1]: pigpiod.service: Main process exited, code=exited, status=1/FAILURE
Jan 31 06:58:55 asciipi systemd[1]: pigpiod.service: Failed with result 'exit-code'.
```

As above (I had to clean up the lock file), I tried a different port but without the -l and it didn't throw an error... and it works

```
pi@asciipi:/webroot/asciipi $ sudo /usr/bin/pigpiod -p 8885
pi@asciipi:/webroot/asciipi $ 
pi@asciipi:/webroot/asciipi $ export PIGPIO_PORT=8885
pi@asciipi:/webroot/asciipi $ pigs pigpv
79
pi@asciipi:/webroot/asciipi $ pigs HWVER
11546898
pi@asciipi:/webroot/asciipi $
```
Since I can't get my software to change its port, I updated the service file to take off the -l and rebooted without a problem and all working now. The process I had to follow to getthis to work out of the box was this:

```
    sudo apt install -y pigpio python3-pigpio python3-gpiozero
    sudo find /lib/systemd/system/pigpiod.service -exec sed -i "s/\/usr\/bin\/pigpiod -l/\/usr\/bin\/pigpiod/g" '{}' \;
    sudo systemctl enable pigpiod
    sudo rm -f /var/run/pigpio.pid
    sudo systemctl start pigpiod
    sudo service pigpiod status
```

closed 2023-02-02 by nigeljohnson73