joan2937/pigpio #560: When, if at all, is it safe to drop root in a daemon that uses pigpio?

Ozzard, 2023-02-03:
There are lots of warnings that pigpio uses /dev/mem, and hence needs to run as root.  That's fine.

I'm building a C daemon that will use pigpio.  I'd rather not keep it running as root if there's an alternative.  Is there a point after which the devices are open, and hence it's possible for the daemon to drop root?  If so, when - directly after `gpioInitialise()`, for example?

Yes, I know that given write access to /dev/mem in the process, this is rather akin to shutting the stable door after the horse has ambled into the field, had a good meal and a nap, looked around, and galloped into the next county.  But let me secure the small things, eh? :-)

guymcswain, 2023-02-04:
If you write your daemon using pigpiod_if you won’t need root privilege.

Ozzard, 2023-02-04:
> If you write your daemon using pigpiod_if you won’t need root privilege.

True.  Not convinced that reduces the attack surface, however - does pigpiod drop root once it's started?  If so, presumably I can look at its approach and do the same.

Given what I'm using this stuff for (factory automation), I'd also prefer to have the smallest possible amount of code to validate and the fewest possible moving parts in the solution.  I'm not a great fan of large machines going haywire on error and someone having to press the Big Red Mushroom Button and clean the mess off the floor :-).