joan2937/pigpio #561: Default systemctl unit file on Bullseye specifies single `-l` option which fails

Paraphraser, 2023-02-13:
I've been helping someone who ran into a problem on a Raspberry Pi 4B after rebuilding their system based on Bullseye.

Before the upgrade, their Node-RED flows using node-red-node-pi-gpiod and pigpiod worked fine. After the upgrade, the GPIO nodes stopped working.

I've been able to reproduce the problem on my own hardware and I believe I've tracked it to the default systemctl unit file for pigpiod.

In and of itself, this is **not** a problem with pigpiod as such, but it *may* have exposed an issue with pigpiod's command line options.

The default systemctl unit file is:

```
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```

The unit is disabled by default on Bullseye. Enabling it results in the daemon exiting with:

```
pigpiod.service: Main process exited, code=exited, status=1/FAILURE
```

If I run the `ExecStart` command in the foreground, I get an error:

```
$ sudo /usr/bin/pigpiod -g -l
2023-02-13 22:23:18 initInitialise: bind to port 8888 failed (Cannot assign requested address)
Can't initialise pigpio library
```

That would seem to suggest port 8888 is already in use, but that simply isn't true:

```
$ sudo netstat -tulpn | grep 8888
$ 
```

The man page defines `-l` as:

```
-l     Disable remote socket interface.  Default enabled
```

The command-line help:

```
-l,         localhost socket only              default local+remote
```

It isn't immediately clear whether `-l` by itself means:

* `localhost:8888` or
* not binding to a socket at all.

The former seems to be the more likely. However, passing localhost explicitly:

```
$ sudo /usr/bin/pigpiod -g -l -n localhost
```

appears to succeed (no error; needs <kbd>control</kbd>+<kbd>C</kbd> to terminate). Running the same command in the background also produces expected behaviour:

```
$ sudo /usr/bin/pigpiod -l -n localhost
$ ps -ax | grep pigpiod | grep -v grep
2933446 ?        SLsl   0:00 /usr/bin/pigpiod -l -n localhost
$ sudo netstat -tulpn | grep 8888
tcp        0      0 127.0.0.1:8888          0.0.0.0:*               LISTEN      2933446/pigpiod
$ sudo killall pigpiod     
```

Conclusion: the daemon launches and binds to localhost:8888.

The material point is that the `-l` option does not seem to work by itself. It's not clear from the documentation whether `-l` *should* work on its lonesome, or whether it always has to be used in conjunction with the `-n` switch.

In other words, I accept that it's entirely possible that pigpiod is behaving correctly, and that 100% of the problem lies with the default systemctl unit file for pigpiod under Bullseye.

I have not drilled any deeper into this to try to understand whether:

1. The Raspberry Pi OS unit file for pigpiod has **always** contained a lone `-l` switch&nbsp;–&nbsp;which would imply a change in behaviour for pigpiod; or
2. Adding the `-l` switch **and** disabling the unit by default are changes introduced in Bullseye, perhaps with the goal of reducing a potential attack surface.

Given the lack of hits when researching this topic, option 2 would seem to be the more likely, but I have no evidence one way or the other.

I'm opening this issue for two reasons:

1. As a heads-up in case a lone `-l` *should* work (ie there really is a bug in pigpiod); and
2. To help anyone else trying to debug GPIO problems in Node-RED flows on Bullseye.

To make node-red-node-pi-gpiod + pigpiod work on Bullseye, you need to do the following:

1. Run:

	```
	$ sudo systemctl edit --full pigpiod
	```

2. Edit the `ExecStart` line to remove the `-l` switch.
3. Save via <kbd>control</kbd>+<kbd>O</kbd>, <kbd>return</kbd>, <kbd>control</kbd>+<kbd>O</kbd>.
4. Run:

	```
	$ sudo systemctl enable pigpiod
	$ sudo reboot
	```
	
Tips:

* Don't be tempted to replace the reboot with a `systemctl start`. If you have any Docker containers running, the Pi will likely hang on the next reboot. I don't know why. 

* Don't forget that Node-RED's "pi gpiod" nodes need to be configured with a Host IP address of the default gateway on the Docker bridge network. It is usually "172.17.0.1":

	```
	$ docker network inspect bridge | jq .[0].IPAM.Config[0].Gateway
	"172.17.0.1"
	``` 

To be absolutely crystal clear, removing the `-l` switch from the `ExecStart` line and enabling the pigpiod unit is all that is needed to allow Node-RED flows running in a Docker container to manipulate GPIO pins on a Raspberry Pi running Bullseye. I've put additional information and logic analyzer traces in [this gist](https://gist.github.com/Paraphraser/bae3d41b60343608a256e1ce62153ff7).

It's *possible* that this is related to #275 #519 and #554

<hr>

```
$ grep PRETTY /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"

$ uname -a
Linux sec-dev 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

$ pigpiod -v
79

$ apt list pigpiod
Listing... Done
pigpiod/stable,now 1.79-1+rpt1 arm64 [installed,automatic]
pigpiod/stable 1.79-1+rpt1 armhf
```

guymcswain, 2023-02-13:
Given the system is running Node-Red, does this imply a desktop environment?  If so, then I'm fairly confident you have hit a known issue (#554) with pigpiod systemd service on Bullseye with a graphical target.  I really don't think the `-l` option solved any issue because that has been in existence prior to Bullseye - and yes it was made a default for 'security'.

It would be very helpful if you could verify my hypothesis by modifying the service file according to #554.

Paraphraser, 2023-02-14:
If I may, I'd like to deal with this in steps.

> Given the system is running Node-Red, does this imply a desktop environment?

To be honest, I don't know how to answer this and be sure I'm actually responding to your question.

My starting image for the test system can be found at:

* [downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2022-09-26/](http://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2022-09-26/)

and, from there, the specific file:

* `2022-09-22-raspios-bullseye-arm64.img.xz`

The `.sha256` in that directory corresponds with the image circled below:

[image]

So, in that sense, I started from the "Desktop" image.

However, I have no interest in the actual desktop environment. My [build scripts](https://github.com/Paraphraser/PiBuilder/blob/d231fbdebbe027a0928a4c8f7c39e08ff2da5ce4/boot/scripts/01_setup.sh#L144) include:

```
echo "Setting boot behaviour to console (no GUI)"
sudo raspi-config nonint do_boot_behaviour B1
```

and VNC is not enabled either.

All my interaction is via SSH terminal window. The HDMI ports have nothing connected. No USB mouse. No USB keyboard.

When I interact with Node-RED in the sense of using its graphical interface, I'm doing that in a browser on a separate machine (Safari on macOS), as in typing `http://raspberrypi.local:1880` into the URL bar.

As far as I'm concerned, I'm not running any Desktop on the Raspberry Pi, so I think the correct answer to your question is, "no, I'm not running a desktop *environment*." The only reason I start from the 64-bit "with Desktop" image (rather than the "Lite" image) is that it saves a bunch of `apt install` commands when I build a system.

> If so, then I'm fairly confident you have hit a known issue (#554) with pigpiod systemd service on Bullseye with a graphical target.

I'm not going to rule that out and I will check your hypothesis but, responding to that sentence on its terms, I don't think there's a "graphical target" involved.

I'd like to go back one or two steps and re-paint the picture:

* 4GB Raspberry Pi 4 running 64-bit Raspberry Pi OS
* Docker
* Node-RED running in a Docker container (non-host mode)
* Node-RED has [`node-red-node-pi-gpiod`](https://flows.nodered.org/node/node-red-node-pi-gpiod) installed
* A flow is running in the Node-RED environment.
* The flow makes use of a `pi gpiod out` node, configured like this:

	[image]

There's a JSON test file [here](https://gist.github.com/Paraphraser/bae3d41b60343608a256e1ce62153ff7#practical-test). All the flow does is tell that `pi gpiod out` to toggle GPIO21 a few times - and I monitor the result on the pin with a logic analyzer.
	
To work (manipulate the actual GPIO21), the `pi gpiod out` node running *inside* container-space needs to be able to communicate with `pigpiod` running *outside* container-space on the host system.

In terms of what I'm trying to achieve, my definition of:

* "success" is I can see alternating highs and lows on the logic analyzer;
* "failure" is the logic analyzer flatlines.

That is 100% of the scope of my interest. Whatever other capabilities `pigpiod` might offer in different situations are out-of-scope.

That's why I say that I don't think there's any "graphical target" involved.

I hope that makes sense.

guymcswain, 2023-02-14:
Sorry that I was sloppy with my question.  Yes, I meant the desktop os.  It has a graphical target, whether you are using it or not, that pigpiod is racing to set up internal SOC registers (my theory).  With the change to the service file the race is eliminated.

Paraphraser, 2023-02-14:
> It would be very helpful if you could verify my hypothesis by modifying the service file according to #554.

There's a lot of stuff in #554 but I take you to mean [this post](https://github.com/joan2937/pigpio/issues/554#issuecomment-1405364041).

> If I've misunderstood, please clarify.

I will start with a baseline situation.

## test 1 - baseline

Confirm pigpiod is not running:

```
$ sudo systemctl is-active pigpiod
inactive
```

Revert to factory conditions:

```
$ sudo systemctl revert pigpiod

$ sudo systemctl cat pigpiod
# /lib/systemd/system/pigpiod.service
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```

Launch based on factory conditions:

```
$ sudo systemctl start pigpiod

$ sudo systemctl is-active pigpiod
failed

$ ps -ax | grep pigpiod | grep -v grep

$ sudo systemctl status pigpiod
● pigpiod.service - Daemon required to control GPIO pins via pigpio
     Loaded: loaded (/lib/systemd/system/pigpiod.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-02-14 14:50:32 AEDT; 13s ago
    Process: 3016 ExecStart=/usr/bin/pigpiod -l (code=exited, status=0/SUCCESS)
   Main PID: 3017 (code=exited, status=1/FAILURE)
        CPU: 75ms

Feb 14 14:50:32 tri-dev systemd[1]: pigpiod.service: Main process exited, code=exited, status=1/FAILURE
Feb 14 14:50:32 tri-dev systemd[1]: pigpiod.service: Failed with result 'exit-code'.

$ sudo systemctl reset-failed pigpiod
```

Result:

* test fails and, for the reasons explained in my original post, my belief is that `-l` is the cause of that failure.

But let's put a pin in that and move on.

## test 2 - your hypothesis

Edit the unit definition so include the lines you want me to test:

```
$ sudo systemctl revert pigpiod
[... 167 lines cut ...]
      Tasks: 4 (limit: 4163)
     Memory: 440.0K
        CPU: 1.257s
     CGroup: /system.slice/pigpiod.service
             └─3239 /usr/bin/pigpiod
```

Result:

* test succeeds. Compared with *my* hypothesis, blending with *your* hypothesis makes no difference to the outcome.

### Squaring the circle

Bring up the Node-RED container:

```
$ cd IOTstack/
pi@tri-dev:~/IOTstack$ docker-compose up -d nodered
[+] Running 1/1
 ⠿ Container nodered  Started                                                                                                                              0.9s
```

Check the GUI:

[image]

Paraphraser, 2023-02-14:
> Yes, I meant the desktop os. It has a graphical target, whether you are using it or not, that pigpiod is racing to set up internal SOC registers (my theory). With the change to the service file the race is eliminated.

I'm not getting any sense of a "race condition". I am just observing a change in behaviour:

* with `-l` - the default - fails
* without `-l` - my proposal - succeeds

And I'll repeat that my proposal applies to the very specific situation described earlier: Node-RED in a Docker container trying to manipulate a GPIO pin.

As I demonstrated before, assuming pigpiod is **not** running via systemctl, if I launch pigpiod in the foreground:

* **with** `-l` then it complains it can't bind to port 8888, even though it can be shown 8888 is not in use;
* **without** `-l` then it binds to 8888 and stays running until terminated.

Even though the information (stderr) seems to get lost, I have no reason to think the failure to launch under systemctl **with** `-l` is occurring for any reason other than the inability to bind to 8888.

When pigpiod is running **without** `-l` then it responds to instructions from a gpio-out Node running in Node-RED in a Docker container.

To me, race conditions generally block access to resources - yes?

The only "resources" I see here are port 8888 and the GPIO. While the presence of `-l` and an inability to bind to 8888 *might* be due to a race condition, I'm not sure how the same race condition is averted in the absence of `-l`.

Similarly, the presence of `-l` results in a crash so there's no possible contention with the GPIO, while the absence of `-l` affords access to the GPIO. Either way there's no inherent race condition.

The only part of this I can't really explain goes to the question of how long `-l` has been the default and whether the problem I'm reporting has been exposed by 64-bit Bullseye. Figuring that out would need me to see whether the behaviour changed were I to do any of the following:

1. Build a Bullseye 64-bit system based on "Lite".
2. Build a Bullseye 32-bit system based on "with desktop".
3. Build a Bullseye 32-bit system based on "Lite".
4. Build a Buster 32-bit system based on "with desktop".
5. Build a Buster 32-bit system based on "Lite".

That's a lot of work and, to be frank, I'm not *quite* that interested in the problem. As I said right at the get-go, my motivation was helping someone else who reported difficulties after migrating to Bullseye. I don't use pigpiod for any of my own projects so I have no "I really need to understand this" imperative that would lead me to try to eliminate build versions and variants.

On the other hand, I'm quite happy to run any other tests you'd like me to try, providing it occurs in my current environment (64-bit Bullseye based on "with Desktop"), and you understand that I don't have any experience with any of the test tools that might be used with pigpiod (ie step-by-step instructions designed for the newbie would be greatly appreciated).

Paraphraser, 2023-02-15:
I think I've figured out the origin of `-l` or, more precisely, the absence of the problem caused by the *assumed* presence of that switch in the target environment ([SensorsIot/IOTstack](https://github.com/SensorsIot/IOTstack)).

The IOTstack [Node-RED documentation](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Node-RED.md#gpio-access--accessgpio-) instructs users to initiate pigpiod from `/etc/rc.local`. Those instructions omit the `-l` switch. Not being a regular user of Pi GPIO pins myself, my first thought was "is pigpiod running?", which led me to systemctl, which led me down this rabbit hole.

I've asked the person I was helping to check to see if the old system had a line in rc.local. I haven't heard back yet but it makes sense in terms of the behavioural change between that person's old/new systems. It also explains why this hasn't come up before in either an IOTstack issue or a question on Discord.

<hr>

I'm not sure whether to close this issue.

On the one hand, I can "solve" the problem in the target environment by changing the IOTstack Node-RED documentation to explain how to use systemctl to set things up properly - for IOTstack.

On the other hand, it seems to me there's still a basic problem that someone coming to a clean Raspbian install and telling systemctl to enable pigpiod gets something that crashes.

It would be different if `-l` worked but wasn't particularly useful without additional configuration to make it fit for some intended purpose. It's the fact that it crashes out-of-the-box that bothers me.

Any thoughts?