joan2937/pigpio #562: Error on arm64 system

hbtecSZE, 2023-02-15:
Hi

I try to get this running on a **Raspberry Pi 4** with **Debian 11.6 64bit** (5.10.0-20-arm64).
I followed the steps on https://abyz.me.uk/rpi/pigpio/download.html.

When I try to start the Daemon with `sudo pigpiod` I get the following error:

`2023-02-15 14:09:28 initPeripherals: mmap gpio failed (Operation not permitted)
Can't initialise pigpio library`

I found the issue https://github.com/joan2937/pigpio/issues/259 which sounded very similar to my problem.
Even tho I use version 79 this problem persists.

I also try to add `iomem=relaxed strict-devmem=0` to `my /boot/firmware/cmdline.txt` file and after a reboot and `sudo pigpiod`, I got the error:

`2023-02-15 14:30:16 initAllocDMAMem: mbox open failed(No such device or address)
Can't initialise pigpio library`

So it looks like there is a problem with the memory handling on an arm64 system?
Is there something I missed, or is it not possible to run this on my system?

Thanks

guymcswain, 2023-02-15:
I can't think of a reason why it is failing.  Is this a fresh install of the OS?  If so, you already had v79 included in the distro.  If you want the very latest you need to:
```bash
wget https://github.com/joan2937/pigpio/archive/refs/heads/develop.zip
unzip develop.zip
cd pigpio-develop
make
sudo make install
```
... although I don't think this is likely to fix your problem, but worth a try.

hbtecSZE, 2023-02-16:
It is a fresh installation of the OS, but I already set up and installed some additional Software.
Maybe also to Mention: I have an RTC already running on the GPIOs.
I had the same Situation with Raspbian, and it worked without problems.

I will try the develop Version later and keep you updated.

hbtecSZE, 2023-03-02:
Apparently the issue is that pigiod is using an interface (/dev/vcio) which doesn't exist on Debian.

Marsu31, 2023-05-23:
I used gdb to trace the failure of the _pigpiod_ command with arguments _-a1 -t0 -g_ :
```
Program received signal SIGSEGV, Segmentation fault.
0x0000fffff7eeb46c in initPWM (bits=10) at pigpio.c:7901
7901	   dmaIVirt[0]->periphData = 1;
```
Hope it helps !