joan2937/pigpio #571: pigpiod on RPI3B+ problems

gratefulfrog, 2023-04-17:
Hello,

I've been using this library for several weeks on an RPI4B+ with no issues whatsoever. It is really a great tool!

I am now trying to run it on an RPI3B+.

raspian 0S 64bit is fully updated, upgraded.

When I setup the pigpiod to startup automatically with sudo systemctl enable pigpiod it does startup but starts continually blinking certain gpio pins. In particular, the ones I need for my project... GPIO 17, 22, 23, 24, 27

If I run sudo systemctl stop pigpiod then restart, it works fine, the gpios stay at 0.

However, in python3:

$ sudo systemctl status pigpiod
● pigpiod.service - Daemon required to control GPIO pins via pigpio
     Loaded: loaded (/lib/systemd/system/pigpiod.service; disabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-04-17 17:32:13 CEST; 5min ago
    Process: 2305 ExecStart=/usr/bin/pigpiod -l (code=exited, status=0/SUCCESS)
   Main PID: 2306 (pigpiod)
      Tasks: 4 (limit: 719)
        CPU: 19.585s
     CGroup: /system.slice/pigpiod.service
             └─2306 /usr/bin/pigpiod -l

Apr 17 17:32:13 rpi3Bplus64bit00 systemd[1]: Starting Daemon required to control GPIO pins via pigpio...
Apr 17 17:32:13 rpi3Bplus64bit00 systemd[1]: Started Daemon required to control GPIO pins via pigpio.

$ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pigpio
>>> pig = pigpio.pi()
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at soft(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
>>> pig.connected
False
>>> 

I am looking for at least a workaround, if possible, to be able to continue my project which requires fully automatic startup.

I am fully available to help debug this issue.

Thanks for any suggestion.
Cheers,
Bob

gratefulfrog, 2023-04-17:
adding to previous, if I start python as root, ie sudo python, I can access pigpio.pi()...

So there are 2 issues:
- flashing gpio pins at startup
- accessing python interface as non root user.
Thanks for any thoughts!

mkende, 2023-11-15:
I’m facing that issue too, usin an IQAudio Dac Pro on an RPi 3B+ (using the I2S output). However the fix of issue #554 does not seem to work for me.

As soon as I start the pigpiod server with `sudo systemctl start pigpiod`  the sound output of the DAC becomes garbled (same happens if I enable the service and restart the RPi).

I’m using the same Raspbian based on Debian 11 32 bit as in that other issue, except that I’m using the Lite OS.

I have tried editing the `/etc/systemd/system/default.target.wants/pigpiod.service` file as per the comment in the other issue, but the issue still persist (whether I activate the daemon manually or whether I enable it and reboot the Pi).

Interestingly, my issue with the I2S output persist even if I stop the daemon. And I need to restart the board for it to work correctly again.

guymcswain, 2023-11-16:
@mkende , if need to use audio with pigpio you'll need to give up some of the functionality.  By default, the I2S peripheral is used for timing.  There is an option to run the daemon using the PWM peripheral instead.  [Read about pigpio daemon options here.](http://abyz.me.uk/rpi/pigpio/pigpiod.html#:~:text=Default%20PCM.%20pigpio%20uses%20one%20or%20both%20of%20PCM%20and%20PWM.%20If%20PCM%20is%20used%20then%20PWM%20is%20available%20for%20audio.%20If%20PWM%20is%20used%20then%20PCM%20is%20available%20for%20audio.%20If%20waves%20or%20hardware%20PWM%20are%20used%20neither%20PWM%20nor%20PCM%20will%20be%20available%20for%20audio.)

mkende, 2023-11-16:
Thanks for the pointer. Do you know if I can still use one of the PWM channels through pigpiod (well, one output of one channel), if I switch the daemon to using PWM timing?

It's probably not worth opening an issue for the second part of the problem as I wouldn't expect the I2S to work now that I have this additional information.

guymcswain, 2023-11-16:
> Do you know if I can still use one of the PWM channels through pigpiod (well, one output of one channel), if I switch the daemon to using PWM timing?

Sorry but no, neither channel will be available for pwm.