joan2937/pigpio #575: more detailed documentation required about gpioWaveCreatePad

audi0615, 2023-05-07:
Hi.

I understand that gpioWaveCreatePad has been newly added for continuous filling and transmitting loops by padding out waves to have the same size for the re-use of resources.

However, I believe that a more detailed explanation is needed for the following:

Q1. Can wave chunks for filling and transmitting have different delays and numbers of pulses?

Q2. How do you determine pctCB and pctBOOL?

```
pctCB: 0-100, the percent of all DMA control blocks to consume.
pctBOOL: 0-100, percent On-Off-Level (OOL) buffer to consume for wave output.
pctTOOL: 0-100, the percent of OOL buffer to consume for wave input (flags).
```

In the following example, let's assume that nextWaveChunk has 10,000 pulses. Would setting pctCB and pctBOOL to 50% be sufficient? How can I determine the appropriate values for these parameters?

```
 // get firstWaveChunk, somehow
  gpioWaveAddGeneric(firstWaveChunk);
  wid = gpioWaveCreatePad(50, 50, 0);
  gpioWaveTxSend(wid, PI_WAVE_MODE_ONE_SHOT);
  // get nextWaveChunk

  while (nextWaveChunk) {
     gpioWaveAddGeneric(nextWaveChunk);
     nextWid = gpioWaveCreatePad(50, 50, 0);
     gpioWaveTxSend(nextWid, PI_WAVE_MODE_ONE_SHOT_SYNC);
     while(gpioWaveTxAt() == wid) time_sleep(0.1);
     gpioWaveDelete(wid);
     wid = nextWid;
     // get nextWaveChunk
  }
```

Q3. How can I delete all resources in the last loop?

During each loop, a resource that was flagged for deletion in the previous loop is reused. How can I delete the resource in the final loop?

guymcswain, 2023-05-07:
Q1:  Different delays?  Yes, that was the original intention, as I recall.  You could change the number of pulses as long as the chunk fit within the allocated resources.

Q2:  In general, 50% is the logical choice - it splits all available resources so that the next chunk can be computed while the current chunk is transmitting.  If your chunk of 10000 pulses does not fit within 50%, you will need to adjust your chunk size.

Why this parameter?  There was some concern about exhausting  one of the “OOL”s when doing bit-banged SPI operation simultaneously (if I recall correctly).  That situation was never tested as it seemed to be a fairly unusual use case.

audi0615, 2023-05-07:
Thank you guymcswain for your reply. 
regarding Q2, how do i know a certain number of pulses are fit within 50% of the available resources?
Is this related to the hard limit of number of pulses(12,000)? so 6,000 pulses would be ok with 50%?

guymcswain, 2023-05-07:
It’s a bit more complicated than hard number limits so you need to use the API’s that give you the available CB, and other statistics.  After you create the first chunk you can verify you are within 50% of the total.  I suppose a more complete example including this would have improved the doc.

 A good strategy is to pick a chunk size that provides enough total delay to assure you have time to compute the next before the transmission of the previous chunk expires.  In the use cases I originally considered, I was never anywhere close to needing 50% of the available resources.  But, one can never imagine all applications.

guymcswain, 2023-05-07:
I just remembered that a wave in pigpio is really a container for multiple “wavelets” - or waves on different gpio.  Each additional wavelet will consume more resource.

audi0615, 2023-05-07:
Thank you so much guymcswain.
Now everything is very clear.
I would start with 3,000 pulses for each wave chunk to control 3 stepper motors simultaneously.
 
Thanks again.

closed 2023-05-18 by audi0615