joan2937/pigpio #580: general on pigpiod daemon

ss555, 2023-07-04:
sudo pigpiod -s 2 -b 200 -f,
what is the influence of buffer size?

I need to run PWM command every 2-5ms, is 200ms enough?

guymcswain, 2023-07-04:
The PWM commands affect gpio outputs while -s -b parameters relate to gpio input sampling.  So, to answer your question as written, there is no influence.   But I suspect that I'm not understanding your question entirely.

ss555, 2023-07-04:
ok thanks, but what is the use of buffer when reading an input? Is it RAM allocated to the activity? we can't make it dynamic or it's too slow?

guymcswain, 2023-07-04:
You really don’t need to change anything from the defaults.  Are you having an issue?

ss555, 2023-07-06:
I have another setup with 2 digital inputs, that I want to sample from as fast as possible, but if it is not regular it's not worth to go fast, so I wanted to play with sampling rate and buf size

closed 2023-10-08 by guymcswain