joan2937/pigpio #582: Can create SPI wave in nano seconds resolution？

andyzhuang2015, 2023-08-19:
I use ADS8688A with the 500kSPS. It need 64 clock time per sample. However, the pigpio library use 1us as the min clock time. The max speed I can achieved is (1000 / 64) kSPS, which is much smaller than 500. 

So how can modify the SPI clock resolution to 10ns  or 100ns ?

guymcswain, 2023-08-19:
The hardware SPI supports a baud rate maximum of 32M.

andyzhuang2015, 2023-08-19:
> The hardware SPI supports a baud rate maximum of 32M.

Yes. But when use `rawWaveAddSPI` ,  the parameter's `rawSPI_t`'s `clk_us` min value is 1us. This limit the clock frequency to 1M. Any workaround to improve the `Wave` frequency? 
@guymcswain This blocking me too much time. Will be ery appreciate for your help!

guymcswain, 2023-08-19:
The construction of output waveforms is limited to 1 microsecond timing.  Ie, the fastest square waveform you can generate is  500 KHz.

andyzhuang2015, 2023-08-19:
> The construction of output waveforms is limited to 1 microsecond timing. Ie, the fastest square waveform you can generate is 500 KHz.

Is it possible to improve the waveforms' CLK timing to nano seconds?

guymcswain, 2023-08-20:
No.  Waveforms are controlled by DMA.  To try to make it run sub microseconds would be futile as that would create excessive bus traffic interfering with normal activity.

Why do you not use the hardware SPI peripheral which is intended for such purposes?

andyzhuang2015, 2023-08-21:
@guymcswain  I use RPI to do this because it need no other devices and I saw that the clocks can achieve 100MHz or higher for RPI peripheral. So I want to test the max ablity of the speed.
@guymcswain  You said `hardware SPI peripheral`, did you mean using FPGA DAQ card, like NI Corporation's?

guymcswain, 2023-08-22:
RPi is made up of many peripherals.  The 100MHz rate you mention is only achievable in certain Gpio pins via the Clock peripheral.  When I mentioned the SPI peripheral I am referring to the APIs: ` spiOpen, spiTransfer, spiClose`.

andyzhuang2015, 2023-08-22:
@guymcswain Thanks for your patience.  I need to monitor the signal continuously without time gap between two data acquest. I tried `spiOpen`, `spiXfer` (I didn't found `spiTransfer` API), `spiClose`. But there is a dubious time gap between two `spiXfer` invoking. I think the System schedule to do other things in the time gap. Is there any workaround to avoid that?

andyzhuang2015, 2023-08-27:
@guymcswain If I want to streaming the input data without andy gaps, can I use `spiXfer` to achieve this?