joan2937/pigpio #586: pigpio probably won't work on Pi5 (or later Pi's)

joan2937, 2023-09-28:


guymcswain, 2023-09-28:
Without specifications we can never know.  Please link any documentation as it becomes available:  BCM2712, RP1, other?

FilipSLO, 2023-10-11:
This is very sad. But that draft is improving so maybe all is not lost.

deanfourie1, 2024-01-05:
So i'm assuming this is a no go on the RPI5?

Trying to launch pigpiod with

`sudo pigpiod`

I get 

```
w
```

guymcswain, 2024-01-09:
Correct.

spacebares, 2025-06-25:
then what hope is left for non root gpio access in ubuntu ???

andrewdavidmackenzie, 2025-09-25:
"probably" - has anyone actually tried it on Pi5?

Without reading the entire PDF, can you explain why you think it is broken on the Pi5?

ozel, 2025-09-26:
the SMI interface was removed in RPi5 which enabled pigpio being that nicely performant and low latency. Another project, CaribouLite SDR hat has the same issue. They plan to circumvent it with different approaches, the most promising seems the PIO feature of the new RP1 I/O companion chip: https://github.com/cariboulabs/cariboulite/issues/180#issuecomment-2067771275

There's now an official PIOlib for talking to the RP1 on a RPi5, but there are limitations for streaming and low latency I/O. I'm not sure it could compete with what's possible on the RPi4 via SMI. Has anyone tried PIOlib? 

https://www.raspberrypi.com/news/piolib-a-userspace-library-for-pio-control/

I guess the best bet would be waiting for the next Pi, which likely gets an upgraded RP1 that hopefully will have a dedicated interface channel to the linux CPU only for doing real-time I/O stuff - similar to what's possible on a BeagleBone board with PRUs since ~10 years...