joan2937/pigpio #587: Allow get/set of slew rate and hyst bits for gpio

noafterglow, 2023-10-06:
I would like to change the bank drive speeds but have been unable to follow the spaghetti all the way to where it actually gets set in the register.  Note that the actual hardware register does get written since drive strength is there and we have those functions already, but those slew rate and hyst bits are not exposed.

guymcswain, 2023-10-07:
Leave the spaghetti, take the cannoli.
`pigs pads 0 16 // gpio 0-27 set to drive strength 16 mA`

[documentation](http://abyz.me.uk/rpi/pigpio/pigs.html#:~:text=bad%C2%A0pad%C2%A0number-,PADS,-pad%20padma%20%2D%20Set)

guymcswain, 2023-10-07:
Ok, you have the drive working.  Yeah, I don't think the slew rate, hysteresis were implemented.

guymcswain, 2023-10-07:
[maybe this answers slew rate](https://forums.raspberrypi.com/viewtopic.php?p=1019408&hilit=slew+rate#p1019408)