joan2937/pigpio #589: pigpio will not run on a Pi 5

JinShil, 2023-11-07:
pigpio is hard-coded to not run on a Pi5.  See https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/pigpio.c#L7353-L7356

Do the maintainers plan to update pigpio to support the Pi5?  Will this repository entertain pull requests to port pigpio to the pi6?

If not, please update the README and/or archive this repository to let users know.

guymcswain, 2023-11-07:
> or archive this repository to let users know

... seems fairly rash given the installed base of a bazillion working devices vs the literal handful of pi5.

> Do the maintainers plan to update pigpio to support the Pi5?

I have the aspiration to do so but my first quick read of the available docs suggest this will be a huge task, if not impossible without breaking compatibility.

> Will this repository entertain pull requests to port pigpio to the pi~6~5?

Given the potentially monumental task, I think it would be wise to lean into the community for help.  At a minimum, we would want to retain architectural oversight of the project.

sworteu, 2023-11-07:
@guymcswain what would be so different between thos pi's 4 and 4 ? Isn't the pi meant to be backwards compatible?

guymcswain, 2023-11-08:
@sworteu pi4 is supported.  You need to open a separate issue and give details around the failure.

> Isn't the pi meant to be backwards compatible?

Yes but this meaning ends at the os level.  pigpio touches the metal.

sworteu, 2023-11-08:
i tried libgpiod and that seems to work..;) 
https://kernel.googlesource.com/pub/scm/libs/libgpiod/libgpiod/+/v0.2.x/README.md
installed trough: sudo apt install gpiod
Note this is not pigpiod but gpiod.

ebaauw, 2023-11-08:
Or more precisely, the code for the BCM2712.
https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/pigpio.c#L13791-L13830

I would create a PR, but I don't know what values to provide.  I suspect the BCM2711 values won't work, now that GPIO is handled by the RP1 southbridge?

guymcswain, 2023-11-08:
pi5 is not supported by pigpio.  pi5 is built with a new chip architecture that is incompatible with previous iterations.

huesla, 2023-11-12:
Some thoughts of a user:
Perhaps one could evaluate which features still work on the Pi5 and which do not, and then create a slimmed-down PiGpio version just for the Pi5, to which the (still) missing features are gradually added. This would destroy backwards compatibility, but it would allow users to use still working features (if available) right away and at the same time create a basis for the future.
Unfortunately, I do not have the expertise to support in the development. I am using PiGpio only, especially I2C and Wave Pulsing.

pelwell, 2023-11-12:
Although I recommend the use of libgpiod, for direct hardware access you may find the RP1 support in pinctrl (https://github.com/raspberrypi/utils) useful as a starting point.

guymcswain, 2023-11-14:
> ... create a slimmed-down PiGpio version just for the Pi5, to which the (still) missing features are gradually added.

I would generally agree with this strategy, however, there are features/capabilities of pigpio that distinguish it from all other gpio libraries:  the DMA read engine for sampling gpio and the DMA write engine for pwm, servo and custom waveforms.  I have not taken the time to look into it yet but I'm hoping there exists compatible interfaces on Pi5 such as the VPU mailbox registers allowing the DMA peripheral to be memory mapped.  If a more kernel friendly approach is available, that should be explored as well.

Brunnian, 2024-02-25:
I have just experienced this problem too

> gpioHardwareRevision: unknown rev code (d04170)

so  I am currently snookered

samskiter, 2024-03-25:
Is there a beer/coffee fund to which one could contribute to support RPi5 development? Appreciate it's a monumental task, but this is an excellent library and it would be a shame to see the ecosystem of RPi GPIO libraries grow ever larger!

[image]

alessandromrc, 2024-05-05:
Between this repo https://github.com/G33KatWork/RP1-Reverse-Engineering and the pinctrl [source](https://github.com/raspberrypi/utils/tree/master/pinctrl), it shouldn't be too hard to implement basic GPIO features... on the other hand, re-implementing servo control etc can become a little harder...

SonoraTechnical, 2024-05-31:
IS rpi-lgpio a near drop in replacement for basic functionality?
Can we do this?
sudo apt remove python3-rpi.gpio
pip3 install --break-system-packages rpi-lgpio

guymcswain, 2024-06-26:
> Is there a beer/coffee fund to which one could contribute to support RPi5 development? Appreciate it's a monumental task, but this is an excellent library and it would be a shame to see the ecosystem of RPi GPIO libraries grow ever larger!

Beer and coffee are not the problem.  I'd do it for free if the Raspberry Pi Foundation would publish APIs to access the internal peripherals - like DMA in particular - and memory mapping IO.  All of this was done using a mailbox interface to the VC processor in generations 1-4.  That's all changed in gen 5.  Thus, a new standard (architecture).

guymcswain, 2024-06-26:
Unfortunately, I don't have any working relationship or contacts within the foundation.  The author of this library, @joan2937, perhaps does.

pelwell, 2024-06-26:
1. It's not the Foundation - it's the engineering company Raspberry Pi Ltd. that you would need to talk to, but...
2. I don't know which API you are referring to. Most of pigpio does direct accesses to the hardware from userspace. The datasheet for RP1 (responsible for the user-facing GPIOs on Pi 5) is available, as is the source code for pinctrl - the replacement for raspi-gpio that also does direct hardware accesses.
3. I don't think you should ping Joan - if they wanted to be involved with Pi 5 they would have by now.

guymcswain, 2024-06-26:
Thanks Phil, the problem is not the gpio control but all of the DMA control that is at the core of pigpio.  All that was done through facilities known as mailbox registers via the video core processor.  All is that is gone with pi5.

It will be a huge lift to rewrite the core part of this library and a non starter if there is no assurance that it changes again in the future.

People smarter than me wrote this code.  I’d be happy to take a look if someone could point me to documentation that allows the same functionality on the pi5.

pelwell, 2024-06-27:
With Pi 5 the role of the firmware has reduced. Apart from some basic power and clocks management, its job is basically to load the kernel and get out of the way. Having said that, I think it's only the memory allocation that is performed by the firmware - there is no "DMA this data to here" API. Since RP1 has access to all of system RAM there is no need for a special allocator.

AFAIK we aren't permitted to publish the documentation for the Synopsys DMA controller, but the Linux driver is available as executable documentation: https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/dma/dw-axi-dmac/dw-axi-dmac-platform.c

JinShil, 2024-06-27:
@pelwell I think the DMA feature that is being referred to here is the clever usage of it to achieve hardware timed PWM.  See https://stackoverflow.com/a/53033272/539972

PIO is supposed to come to the Pi5, right?  If so, I think that would make pigpio less relevant for the Pi5, or at least provide a way to implement the features that pigpio currently provides for the Pi4 and earlier.

John-Smiths, 2024-07-18:
在 py 中使用gpiod

```
apt update && apt upgrade && pip install gpiod && apt install libgpiod-dev
```

TobiasReich, 2025-07-27:
I assume there are no news regarding this? I was hoping - after a year - maybe someone smart has found a solution for this.

samskiter, 2025-07-28:
> Porting pigpio to the Pi 5 wouldn't make much sense as the architecture is
> heavily different than the previous models. Also, given the alternatives
> that already exist, there would be no benefit on maintaining pigpio for it.

What are the best alternatives in your opinion?