joan2937/pigpio #596: Is there a way to flush an open gpioSerialRead buffer?

davthomaspilot, 2024-01-20:
Or, better yet, a way to "pause" the accumulation of timestamped edges that are being accumulated in the circular buffer?

I'm using serial read to get data from an RF receiver.  I'm adding functionality for the RF receiver to scan multiple channels.  Whenever the frequency is changed, the serial data being read will be invalid for at least a known period of time.

I'd like a way to "flush" the read buffer (quickly) and/or the ability to stop/start edge acquistion.

I'm already using a "hacked" version of gpioSerialRead to return the timestamp data.  So, maybe I can modify that to get what I want.  But, maybe there's a "built-in" way to get what I need?

guymcswain, 2024-01-20:
Things you may have already thought about:
- Closing and then re-opening gpioSerialData.
- Reading all the data (8196 bytes) and then discarding the data.

Other ideas:
- Use a glitch filter to filter noise frequencies above the bit rate.
- Decode the serial data externally to pigpio using the raw events data.  In your serial decoder you can search for the RF preamble (mentioned in #595).

jw3, 2024-01-20:
> a way to "flush" the read buffer (quickly)

I implemented a "flush" by setting the `readPos` and `writePos` of a `wfRxSerialTs_t` to `0`.

Haven't had any issues with it that I can recall.

davthomaspilot, 2024-01-20:
> _Closing and then re-opening gpioSerialData._

I'll try that!  Do you have a guess on how much time expires between a close and reopen?

> _Reading all the data (8196 bytes) and then discarding the data._

I tried doing a read and discarding the data--but maybe not ALL of it.  Maybe there's more in the buffer than I expect, need to check my math or just do a quick experiment
_

> Use a glitch filter to filter noise frequencies above the bit rate

_.

You mean external hardware?  There are several things I could try if I wanted to add just a bit of hardware, but I haven't given up on trying to do it in code.
`
Decode the serial data externally to pigpio using the raw (level, tick) callbacks. In your serial decoder you can search for the RF preamble (mentioned in https://github.com/joan2937/pigpio/issues/595).`

You mentioned that in the other thread and maybe I'm not "getting it".  But, I don't see how to make it work.

The beginning of a valid message is indicated by a period of  1.06 msec.  And, the serial read needs to start right when that 1.06 msec ends.  So, by the time the 1.06 msec is detected, it's already to late to start the serial read.  So, the implementation I have now reads all the time, but discards unneeded data.

Discarding extra transitions "works", but slows down how fast the receiver can scan multiple channels.  For some reason, when the receiver starts back up after a frequency change, I'm getting many more "false triggers"--the receiver is outputting logic ones for relatively long periods of time.  

Thanks for your suggestions!  A couple of paths to experiment with.

davthomaspilot, 2024-01-20:
>  implemented a "flush" by setting the readPos and writePos of a wfRxSerialTs_t to 0.

That sounds like what I need.  How did you do it, buy adding a function to the pigpio library?

davthomaspilot, 2024-01-20:
What causes gpioSerialRead to return when less characters are in the internal buffer than the specified size of the external buffer?  Is it a timeout?  Can it be configured?

(It stays in the read too long if I try doing a read then discarding as a way to do the flush).

davthomaspilot, 2024-01-20:
Also, closing and reopening seems to take too long, unless doing that introduces something else that breaks things.  Same symptom as doing a dummy serialread than discarding--the entire message is missed.

jw3, 2024-01-20:
> >  implemented a "flush" by setting the readPos and writePos of a wfRxSerialTs_t to 0.
> 
> That sounds like what I need.  How did you do it, by adding a function to the pigpio library?  

Yes

davthomaspilot, 2024-01-20:
Thanks!

Trying it now...

davthomaspilot, 2024-01-20:
Works great!  Thanks!