joan2937/pigpio #601: SPI Slave - BSCX not consistently working RPi4B

karlo922, 2024-02-19:
I have a ESP32 set to an SPI Master with 100kHz, Mode 2 and just want to send 0x19 to the RPi4 running pigpiod and using shell for output.
When I use piscope, everything seems fine for me on the pins:
[image]

But `pigs bscx 0x213` or also `pigs bscx 0x21B` always just returns `1  18` so an empty buffer.

What could be wrong?

guymcswain, 2024-02-19:
BSC MISO is pin 10 on RPI4b according to docs.

karlo922, 2024-02-19:
You mean GPIO10 or Pin10? Where do you find that?
I now tried GPIO10 but this does not change anything apart that piscope now shows that 10-MOSI gets the signal.

guymcswain, 2024-02-19:
GPIO10 or pin number 19.  [Here](https://abyz.me.uk/rpi/pigpio/pigs.html#BSCX)

karlo922, 2024-02-19:
After trying out the given BitBang example which worked, I tried to change my ESP32 so that it always sends twice - now it works.. Thank you, although I still do not really understand the issue.

guymcswain, 2024-02-20:
It's been a while since I've played with the BSC-SPI peripheral.  From the documentation, it appears the LSB of the first byte on MOSI controls whether data is to be written to the BSC-SPI FIFO or read from the BSC-SPI FIFO.  So yes, you need to send two bytes to send one data byte.

> The BSC slave in SPI mode deserializes data from the MOSI pin into its receiver/ FIFO when the LSB of the first byte is a 0. No data is output on the MISO pin. When the LSB of the first byte on MOSI is a 1, the transmitter/FIFO data is serialized onto the MISO pin while all other data on the MOSI pin is ignored.

karlo922, 2024-04-20:
I guess all info is here - we can close it.

closed 2024-04-20 by karlo922