joan2937/pigpio #602: i2c_open() returns the same handle for two  devices on the same bus

wittend, 2024-03-07:
I hope that this is something ignorant on my part, but I have looked for something similar here and on the net, and have not found it.

I am using pigpio_if2 to read and write to sensors connected to  a Raspberry Pi 4, testing using the daemon (pigpiod) or when launched by another program  linked to pigpio which acts as a daemon.

uname -a returns: Linux Grape2-Pi32 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
but I am using the 32bit userland install of Raspbian for continuity reasons.

pigpiod -V returns 79.

I am working with hardware that  presents three devices on the /dev/i2c-1 bus. One appears  at  0x20, and has many registers which I read and write.  The others are two MCP9808 temperature sensors, one local and one remote, which present at 0x18 and 0x19, respectively.  'i2cdetect -y 1' shows all sensors present at the expected addresses.

In general, your wonderful tools are working very well. At first I was just ignoring the local temp sensor.   But when I was asked report both, as I had in earlier (non-pigpio) versions of the software.  But when I was asked to make it report both temperatures, odd behavior began to appear.

Early in the code I open handles on each device using i2c_open().   In a stripped-down test version of the code this is what happens:

 $ ./test_pigpio_opens
    [CHILD] pigpio_start(RM3100) OK. p->pi = 0
    [CHILD] i2c_open(RM3100)    Magnetometer OK. Address: 0x20, Handle: 0
    [CHILD] i2c_open(MCP9808-0) Local Temp OK.   Address: 0x18, Handle: 1
    [CHILD] i2c_open(MCP9808-1) Remote Temp OK.  Address: 0x19, Handle: 1

(This is the same in the actual program)

The handle for the magnetometer is usually some integer, often 0, depending on how long it has been since I restarted pigpiod.
The handles returned for the two MCP9808 sensors are the same, almost always 1.

When my code reads the ambient temperature from these sensors using i2c_read_i2c_block_data() with the appropriate handle 
    i.e: rv = i2c_read_i2c_block_data(p->pi, p->remoteTempHandle, MCP9808_REG_AMBIENT_TEMP, (char *) data, 2);
    i.e: rv = i2c_read_i2c_block_data(p->pi, p->localTempHandle, MCP9808_REG_AMBIENT_TEMP, (char *) data, 2);
each sensor returns the same value.  That is not too surprising, in that the handle that I pass: - remoteTempHandle or localTempHandle, are equal, both both usually 1. 

The output from the actual program looks like this, with both "lt" and "rt" values changing in locstep:
$ ./magdata

[CHILD] In ./magdata child process.
    [CHILD] Open PIPE Out OK.
    [CHILD] Open PIPE In OK.
    [CHILD] Before setting up GPIO.
    [CHILD] i2c_open(RM3100) Magnetometer OK. Reg: 32, Handle: 5
    [CHILD] i2c_open(MCP9808) Local Temp OK. Reg: 24, Handle: 1
    [CHILD] i2c_open(MCP9808) Remote Temp OK. Reg: 25, Remote Handle: 1
    [CHILD]: { "ts":202403 7242728, "lt": 33.12, "rt": 33.12, "x":223.881, "y":-423.296, "z":-93.260 }
    [CHILD]: { "ts":202403 7242729, "lt": 33.12, "rt": 33.12, "x":223.947, "y":-423.339, "z":-93.255 }
    [CHILD]: { "ts":202403 7242730, "lt": 33.12, "rt": 33.12, "x":223.945, "y":-423.322, "z":-93.262 }
    [CHILD]: { "ts":202403 7242731, "lt": 33.06, "rt": 33.06, "x":223.951, "y":-423.322, "z":-93.255 }
    [CHILD]: { "ts":202403 7242732, "lt": 33.12, "rt": 33.12, "x":223.962, "y":-423.317, "z":-93.262 }
    [CHILD]: { "ts":202403 7242733, "lt": 33.12, "rt": 33.12, "x":223.955, "y":-423.311, "z":-93.266 }
    [CHILD]: { "ts":202403 7242734, "lt": 33.06, "rt": 33.06, "x":223.963, "y":-423.312, "z":-93.259 }

Any suggestions?

wittend, 2024-03-07:
Full source code is TL;DR, but available on request.

guymcswain, 2024-03-09:
Is your code running in multiple threads; ie, do you have multiple sockets open to the daemon?

wittend, 2024-03-09:
No, my code (especially my test code) is run in a single thread.  

I do use wait_for_edge() in my main loop to trigger on a rising edge from a GPS/GNSS pulse per second (PPS) input to align my measurements.  But even that is not present in the test code, and these i2c_open() calls  are made before that loop is begun.

wittend, 2024-03-09:
Also, I guess that I didn't answer your question completely.  

I am not aware of any way that I might have more than one socket open to the daemon process.  There are internals of pigpio that I don't completely grok, but I only call pigpio_start() one time, and I assume that this is what opens the socket(s).

guymcswain, 2024-03-09:
> [CHILD] i2c_open(RM3100) Magnetometer OK. Reg: 32, Handle: 5
[CHILD] i2c_open(MCP9808) Local Temp OK. Reg: 24, Handle: 1
[CHILD] i2c_open(MCP9808) Remote Temp OK. Reg: 25, Remote Handle: 1

Please break these down to show the actual parameters supplied to the API.

guymcswain, 2024-03-09:
I'll take a look at this later on this weekend.  If I don't see anything obvious then I would need to replicate the test locally.  I'm not inclined to do this as it will require finding suitable i2c devices and so forth.  I could be persuaded, however, if you can confirm your issue is replicated on a normal OS install.  I'm not sure how your system is configured but it didn't sound like it was the standard image.

wittend, 2024-03-09:
I am running :
------------------
 $ hostnamectl
   Static hostname: Grape2-Pi32
         Icon name: computer
        Machine ID: 7d6c88b41ced4ce484be513a17843c50
           Boot ID: c1936883fcca47c4beb71618a786baf6
  Operating System: Raspbian GNU/Linux 11 (bullseye)
            Kernel: Linux 6.1.21-v8+
      Architecture: arm64
Also:
---------
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

But I (at least) thought that I had used a 32 bit version .  This problem has been replicated on other systems used by the project and I am waiting to hear exactly what  they are working with.  In my case there should be no other major wierdness, as I set up this Pi 4 for building and testing this code.  I don't believe I have done much beyond run occasional apt updates in the last 2-3 months.

guymcswain, 2024-03-10:
Yep, the code is pretty straight forward.   Maybe something to do with the pigpio_if2 client.  Can you try other clients?  I'll take a look at it in the next few days.

wittend, 2024-03-10:
Guy,

I have a python utility that I made to check the overall configuration.  I'll fool with that and see how it behaves.
I'll also try it as a standalone pigpio build. 
We are kind of crushing to get this out to distributed clients before the April 8th eclipse in the US for Doppler-shift experiments.

Thanks,
Dave

guymcswain, 2024-03-10:
Are you running on only pi4 or other versions as well?

wittend, 2024-03-10:
Just Pi-4B.  There are about 40 total measurement stations that will use
this.

Dave

guymcswain, 2024-03-10:
You can just use PIGS:
```
pigs i2co 1 20 0
pigs i2co 1 18 0
pigs i2co 1 29 0
```

wittend, 2024-03-10:
Thanks!.
I will try that

guymcswain, 2024-03-10:
You might also want to turn on more verbose debug level:
```
pigs csi 4
```
Then show the messages from the error pipe:
```
cat /dev/pigerr&
```

wittend, 2024-03-10:
***@***.***:~ $ pigs i2co 1 20 0
19
***@***.***:~ $ 2024-03-10 13:51:11 pthSocketThread: Connection
accepted on socket 5
2024-03-10 13:51:11 pthSocketThread: SO_KEEPALIVE enabled on socket 5

2024-03-10 13:51:11 i2cOpen: i2cBus=1 i2cAddr=20 flags=0x0
2024-03-10 13:51:11 pthSocketThreadHandler: Socket 5 closed
pigs i2co 1 18 0
20

Connection accepted on socket 7
2024-03-10 13:51:27 pthSocketThread: SO_KEEPALIVE enabled on socket 7

2024-03-10 13:51:27 i2cOpen: i2cBus=1 i2cAddr=18 flags=0x0
2024-03-10 13:51:27 pthSocketThreadHandler: Socket 7 closed
21
accepted on socket 5
2024-03-10 13:51:54 pthSocketThread: SO_KEEPALIVE enabled on socket 5

I entered these pigs commands one line at a time.  I'm not sure whether you
meant them to be entered in a block.  Perhaps that way they all get the
same socket?

Dave
2024-03-10 13:51:54 i2cOpen: i2cBus=1 i2cAddr=19 flags=0x0
2024-03-10 13:51:54 pthSocketThreadHandler: Socket 5 closed

guymcswain, 2024-03-10:
So if I'm reading this correctly, the three handles returned were 19, 20 and 21.

wittend, 2024-03-10:
But it looks that way to me., FWIW.
But I really don't have any experience with pigs.
I don't know exactly which interface pigs uses, and what its output means.

Dave

wittend, 2024-03-10:
> Pigs is executing the same API, i2c_open(). The return value is the
> handle.

Does it do it through pigpio_if2 of some other way?

guymcswain, 2024-03-10:
I think I can try your test script on my pi4b locally without and i2c devices.  But I won't be able to do so until tomorrow.

guymcswain, 2024-03-10:
> Does it do it through pigpio_if2 of some other way?

It shouldn't but I need to look at that code as well.

wittend, 2024-03-10:
I'm quite sure that the RM3100 does not support the full SMBus interface.
I'm not quite as sure about the MCP9808s.
I am confused in the pigpio_if2 documentation about which calls are SMBus
calls and which work with older I2C interfaces.  I have been using
i2c_read_i2c_block_data
<https://abyz.me.uk/rpi/pigpio/pdif2.html#i2c_read_i2c_block_data>() to
read registers on the assumption that it sticks to straight I2C and also
because it mostly works. Is that correct?

wittend, 2024-03-10:
I'll take any help I can get.  Really appreciate it.

Dave

guymcswain, 2024-03-10:
I have not played much with all the i2c apis.  Looking at the documentation it does say the SMBus apis are denoted with some text describing the actual bus cycles occuring.  Or something to that affect.  I and believe i2c_read_i2c_block_data is one of them.

guymcswain, 2024-03-10:
Well, I'm not sure about my last statement.  It seems i2c_read_i2c_block_data is standard.  Sorry I'm not an i2c expert.

guymcswain, 2024-03-10:
In any case, it shouldn't matter for the open method.  I think.

guymcswain, 2024-03-11:
I just ran your test script and I get the same failure:
```bash
    [CHILD] pigpio_start(RM3100) OK. p->pi = 0
    [CHILD] i2c_open(RM3100)    Magnetometer OK. Address: 0x20, Handle: 0
    [CHILD] i2c_open(MCP9808-0) Local Temp OK.   Address: 0x18, Handle: 1
    [CHILD] i2c_open(MCP9808-1) Remote Temp OK.  Address: 0x19, Handle: 1
```

guymcswain, 2024-03-11:
If I'm not mistaken, it appears you have a single ctlList structure instantiated but you update the same handle member each time you call ic2_open().  Am I missing something?

wittend, 2024-03-11:
Let me go where I can double check.  I believe that I mean to return a
different handle for each.  Have I gone blind???

Dave

guymcswain, 2024-03-11:
Oops, I think I need more coffee this morning.  They are individual elements of the same structure.

wittend, 2024-03-11:
The code does use a single ctlList structure, but I am updating a
*different* handle element with each call.

wittend, 2024-03-11:
I haven't had mine, either.  The time change over the weekend wreaks havoc
on my head.

guymcswain, 2024-03-11:
Still too early for me to see why, but if I break our the i2c_open() calls from the if statement then it seems to work:
```
//=========================================================================
// test_pigpio_opens.c
//
// A command line interface for the RM3100 3-axis magnetometer from PNI Sensor Corp.
//
// Author:      David Witten, KD0EAG
// Date:        March 8, 2024
// License:     GPL 3.0
// Note:        replaces i2c.c (using file system calls to read(), write(), etc.
//              with calls to pigpio. 
//=========================================================================
#include <pigpiod_if2.h>
#include "main.h"

int initGPIO(volatile ctlList *p);
void termGPIO(volatile ctlList *p);
void showPIGPIOErrMsg(int rv);
struct tm *getUTC();
ctlList ctl;

//---------------------------------------------------------------
//  main()
//---------------------------------------------------------------
int main(int argc, char** argv)
{
//    ctlList ctl;
    ctlList *p = &ctl;
    int     rv = 0;
    
    //-----------------------------------------
    //  Setup magnetometer parameter defaults.
    //-----------------------------------------
    if(p != NULL)
    {
        memset(p, 0, sizeof(ctlList));
    }

    p->pi = 0;
    p->i2cBusNumber     = RASPI_I2C_BUS1;
    p->magHandle        = 0;
    p->magnetometerAddr = RM3100_I2C_ADDRESS;
    p->localTempHandle  = 0;
    p->localTempAddr    = MCP9808_LCL_I2CADDR_DEFAULT;
    p->remoteTempHandle = 0;
    p->remoteTempAddr   = MCP9808_RMT_I2CADDR_DEFAULT;
    p->doBistMask       = 0;
    p->cc_x             = CC_400;
    p->cc_y             = CC_400;
    p->cc_z             = CC_400;
    p->x_gain           = GAIN_150;
    p->y_gain           = GAIN_150;
    p->z_gain           = GAIN_150;
    p->tsMilliseconds   = 0;
    p->TMRCRate         = 0x96;
//    p->Version          = Version;
    p->samplingMode     = POLL;
    p->readBackCCRegs   = FALSE;
    p->CMMSampleRate    = 400;
    p->NOSRegValue      = 60;
    p->DRDYdelay        = 10;
    p->magRevId         = 0x0;
    p->usePipes         = USE_PIPES;
//    p->pipeInPath       = fifoCtrl;
//    p->pipeOutPath      = fifoData;
    p->readBackCCRegs   = FALSE;

    rv = initGPIO(p);
    termGPIO(p);
[... 106 lines cut ...]
    if(now == -1)
    {
        puts("The time() function failed");
    }
    struct tm *ptm = gmtime(&now);
    if(ptm == NULL)
    {
        puts("The gmtime() function failed");
    }
    return ptm;
}

//---------------------------------------------------------------
// void termGPIO(volatile ctlList p)
//---------------------------------------------------------------
void termGPIO(volatile ctlList *p)
{
    // Knock down all of the pigpio setup here.
    p->magHandle = i2c_close(p->pi, p->magHandle);
    p->localTempHandle = i2c_close(p->pi, p->localTempHandle);
    p->remoteTempHandle = i2c_close(p->pi, p->remoteTempHandle);
    pigpio_stop(p->pi);
}

```

wittend, 2024-03-11:
Interesting!  I wonder why that could be?

Dave

wittend, 2024-03-11:
> Verified:
>     [CHILD] i2c_open(MCP9808-1) Remote Temp OK.  Address: 0x19, Handle: 4
> Some illogic in my if statements.
> This may be the answer I was looking (fearing/hoping)  for.  Stupid
> mistakes are usually more easily repaired than systemic ones.

Beginner's C mistakes never quite leave us.
Thanks so much!
Dave

closed 2024-03-11 by wittend