joan2937/pigpio #603: Scheduling repetitive SPI writes

davthomaspilot, 2024-03-11:
I'm considering implementing a frequency scan algorithm that requires precise control of when SPI writes to change frequencies are done.    

If the frequency change is not done quickly enough,  a message on the new frequency will be missed.    So, rather than issuing the SPI commands from code, they are generated from "waves"

Five SPI writes are required to change the receive frequency.  I'd like those writes to occur repetitively, without requiring processor intervention.  The desired result would have the radio hang frequency every X milliseconds (X = somewhere between 5 and 10).

It looks straight forward, but maybe there are issues I'm missing?

Also, the card I'm working with is already wired to the hardware SPI pins.     Can I just do a `sudo dtparam spi=off   and use BB SPI on the hardware SPI pins?

`

guymcswain, 2024-03-11:
> Can I just do a `sudo dtparam spi=off and use BB SPI on the hardware SPI pins?

Yes or better, turn it off by editing /boot/config.txt

davthomaspilot, 2024-03-11:
Yes, but I thought I would do it dynamically, at least until I've sure the BB version is working.  (Currently, the code is using the hardware SPI and I don't want to break that. 

Also, I might want to use the hardware UART to do all the register writes required to initially configure the chip.  Then, disable the hardware UART and start the periodic, BB SPI writes

If I'm only doing SPI writes, is there a reason the baud can't be higher than 250K.  Maybe I have to change the default sampling rate to go higher?

guymcswain, 2024-03-11:
For waves (output) the maximum frequency output is 500KHz.  Sample rate does not affect it.