joan2937/pigpio #604: Isr doesn't work on raspberry os 12.5

cedricboudinet, 2024-05-14:
Hello
I am trying to use interrupts on a raspberry 3b+ with this piece of code:
```
#include <stdio.h> 
#include <stdlib.h>

#include <thread>
#include <signal.h>
#include <sys/time.h>
#include <pigpio.h> 
#include <unistd.h>
const int PA= 4; //broche 7
const int port=13;
void onISR(int gpio, int level, uint32_t tick)
{
	static int st=0;
	gpioWrite(PA, st);
	st=1-st;
}
int main(int argc, char **argv) {
	if (gpioInitialise() == PI_INIT_FAILED) {
		printf("ERROR: Failed to initialize the GPIO interface.\n");
		return 1;
	}
	sleep(1);
	int st = gpioSetISRFunc(port, EITHER_EDGE, 1, onISR);
	printf("gpioSetISRFunc=%d\n", st);
	while (1) {
		time_sleep(100);
	}
	gpioTerminate();
}
```

When I run the code with raspberry os 11, the code runs fine and the isr occurs.
However when using raspberry os 12 (with a new installation), gpioSetISRFunc returns -123 (PI_BAD_ISR_INIT).

Can you help me to fix this?

guymcswain, 2024-05-29:
I've been away for a while.   Are you still seeking help?

cedricboudinet, 2024-05-31:
I gave up using ISR with pigpio, I use libgpiod instead.

But if you have a solution to make it working for bookworm, I will try again.

tomplus, 2024-07-07:
It looks like ISR works via an interface "/sys/class/gpio/" which is already removed :/ 
Ref: https://www.thegoodpenguin.co.uk/blog/stop-using-sys-class-gpio-its-deprecated/