joan2937/pigpio #609: Pigpio stuck on activating

Siggerud, 2024-07-04:
Hey 

I installed pigpio on my raspberry pi, but I can't activate it because it's stuck in activating mode. When I type sudo systemctl status pigpiod I get this message. I'd really like to figure out the solution to this, as it seems like a really useful library.

`pigpiod.service - Pigpio daemon
     Loaded: loaded (/etc/systemd/system/pigpiod.service; enabled; preset: enabled)
     Active: activating (start) since Thu 2024-07-04 20:17:55 CEST; 33s ago
    Process: 3109 ExecStart=/usr/bin/pigpiod (code=exited, status=0/SUCCESS)
      Tasks: 0 (limit: 1517)
        CPU: 16ms
     CGroup: /system.slice/pigpiod.service

Jul 04 20:17:55 raspberrypi systemd[1]: Starting pigpiod.service - Pigpio daemon...
Jul 04 20:17:55 raspberrypi systemd[1]: pigpiod.service: Can't open PID file /run/pigpiod.pid (yet?) after start: >
Jul 04 20:17:55 raspberrypi pigpiod[3110]: 2024-07-04 20:17:55 initMboxBlock: init mbox zaps failed
Jul 04 20:17:55 raspberrypi pigpiod[3110]: Can't initialise pigpio library
`

guymcswain, 2024-07-06:
Is this a fresh install?  Did you previously run systemctl start?  I need more context in order to help.

Also, show the output for `systemctl cat pigpiod`.

Siggerud, 2024-07-06:
> Is this a fresh install? Did you previously run systemctl start? I need more context in order to help.
> 
> Also, show the output for `systemctl cat pigpiod`.

I uninstalled it, and then reinstalled right before my post. I didn't work before the fresh install.

This is my system service file.

[Unit]
Description=Pigpio daemon
After=network.target

[Service]
ExecStart=/usr/bin/pigpiod
ExecStop=/usr/bin/pkill pigpiod
Restart=always
Type=forking
PIDFile=/var/run/pigpiod.pid
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=pigpiod

[Install]
WantedBy=multi-user.target

Siggerud, 2024-07-10:
I see I left out some parts of the log. It's basically this over and over.

Jul 01 19:37:01 raspberrypi systemd[1]: pigpiod.service: start operation timed out. Terminating.
Jul 01 19:37:01 raspberrypi systemd[1]: pigpiod.service: Failed with result 'timeout'.
Jul 01 19:37:01 raspberrypi systemd[1]: Failed to start pigpiod.service - Pigpio daemon.
Jul 01 19:37:02 raspberrypi systemd[1]: pigpiod.service: Scheduled restart job, restart counter is at 2.
Jul 01 19:37:02 raspberrypi systemd[1]: Stopped pigpiod.service - Pigpio daemon.
Jul 01 19:37:02 raspberrypi systemd[1]: Starting pigpiod.service - Pigpio daemon...
Jul 01 19:37:02 raspberrypi systemd[1]: pigpiod.service: Can't open PID file /run/pigpiod.pid (yet?) after start: No such file or directory
Jul 01 19:37:02 raspberrypi pigpiod[1813]: 2024-07-01 19:37:02 initInitialise: Can't lock /var/run/pigpio.pid
Jul 01 19:37:02 raspberrypi pigpiod[1813]: Can't initialise pigpio library

Siggerud, 2024-07-10:
Jesus... I misspelled the file name in the service file. Closing the issue.

closed 2024-07-10 by Siggerud