joan2937/pigpio #611: Missed packets using RaspberryPi and nRF24L01.

b1narygl1tch, 2024-07-15:
Hello!
I'm experimenting with Raspberry Pi Zero 2 W and SPI-connected nRF24L01 module.
I use Python library https://github.com/bjarne-hansen/py-nrf24 which uses pigpio (pigpiod).
In nRF24L01 promiscuous mode I see some data on RX, but I definitely don't see packets from my Logitech mouse.

I checked with other project https://github.com/DigitalSecurity/raspjack which uses RPi.GPIO (https://pypi.org/project/RPi.GPIO/) and spidev (https://pypi.org/project/spidev/). Everything works as expected there.

I analyzed py-nrf24 library that I'm using, experimented with different parameters, compared libs outputs, nRF24L01 registers states, etc. It seems that the problem with pigpio library.

Would be appreciated for hints where to start digging to solve the problem!

P.S.: I don't have a logic analyzer.

guymcswain, 2024-07-15:
Sure.  Let's start with some basic testing.  Since I've never used the Pi Zero 2 W, please run the Python `x_pigpio.py` test to verify the installation and that everything works on this platform.
```bash
wget https://github.com/joan2937/pigpio/raw/master/x_pigpio.py
sudo pigpiod # if not already running
sudo python x_pigpio.py
```
All tests should pass:  "TEST x.y PASS (blah blah)"

guymcswain, 2024-07-15:
Also, please report the header output.  For example, my header is:

> Testing pigpio Python module 1.78
Python 3.9.2 (default, Feb 28 2021, 17:03:44)  [GCC 10.2.1 20210110]
pigpio version 79.
Hardware revision 12595477.

guymcswain, 2024-07-15:
Good.  Can you tell me more about what you did.  In particular why you think it's a problem with the lib.

> I analyzed py-nrf24 library that I'm using, experimented with different parameters, compared libs outputs, nRF24L01 registers states, etc. It seems that the problem with pigpio library.

guymcswain, 2024-07-16:
> I compared registers states for both scripts (mine and raspjack) and they have equal values.

So, does this mean the issue can be closed?

b1narygl1tch, 2024-07-16:
> > I compared registers states for both scripts (mine and raspjack) and they have equal values.
> 
> So, does this mean the issue can be closed?

The problem still exists. Despite setups and debug outputs (registers, SPI speeds, etc) for the both scripts are looking the same, I don't see packets from my Logitech mouse while using py-nrf24 and pigpiod.
You can close the issue. I will use another library for my project.

Thanks a lot!

guymcswain, 2024-07-17:
Yeah, given there's nothing more specific to go by and you've got a working solution, I'll close this.

closed 2024-07-17 by guymcswain