joan2937/pigpio #612: Python 3.11.2 is installed but still getting "Depends: python:any (>= 2.7~) but it is not installable"

ChaosBlades, 2024-07-18:
```
sudo apt install python-pigpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python-pigpio : Depends: python:any (< 2.8) but it is not installable
                 Depends: python:any (>= 2.7~) but it is not installable
E: Unable to correct problems, you have held broken packages.
```
```
python
Python 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```
```
python-pigpio/stable,stable 1.79-1+rpt1 all
  Python module which talks to the pigpio daemon (Python 2)
```

guymcswain, 2024-07-18:
I’m not familiar with the package `python-pigpio`.

ChaosBlades, 2024-07-19:
I was not paying enough attention. There is a Python 2 and Python 3 package. Python 2 package can't be installed when Python 2 isn't installed obviously.

```
pigpiod/stable,now 1.79-1+rpt1 arm64 [installed,automatic]
  Client tools for Raspberry Pi GPIO control

python-pigpio/stable,stable 1.79-1+rpt1 all
  Python module which talks to the pigpio daemon (Python 2)

python3-pigpio/stable,stable,now 1.79-1+rpt1 all [installed]
  Python module which talks to the pigpio daemon (Python 3)
```

closed 2024-07-19 by ChaosBlades