joan2937/pigpio #613: gpioSetISRFunc error -123

TMagerat, 2024-08-05:
Hello

I'm struggling with the function
gpioSetISRFunc
I always have a bad init error -123
I tried to find some solution by disabling the W-1 interface but it does not work 

This is my config.txt

I need this function because I have 2 different square signals on my PIN BCM 17 and 27 that represent the speed of 2 different rotors
I want that each time a RISING_EDGE is occuring it computes the times between the last and the current rising_edge

I tried to do 

```
sudo -s
cd /sys/class/gpio/
echo 17 > export 
cd gpio17
echo in > direction 
echo both > edge 
cd /sys/class/gpio/
echo 17 > unexport 
```

as mentionned here #97  https://github.com/fivdi/pigpio/issues/97

but sadly after the line 
```
echo 17 > export 
```
I have an invalid argument error
My code is something like this :
```
void rotor1_interrupt(int gpio, int level, uint32_t tick) {
    static uint32_t last_tick = 0;
    uint32_t diff_tick;

    
    diff_tick = tick - last_tick;
    last_tick = tick;

    if (diff_tick > 0) {
        rotor1_speed = 1.0 / ((double) diff_tick / 1000000.0);
        clock_gettime(CLOCK_MONOTONIC, &speed1);
        time1 = (speed1.tv_sec - start.tv_sec) + (speed1.tv_nsec - start.tv_nsec) / 1.0e9;
    }
    rotor1_speed_buffer[rotor1_data_index] = rotor1_speed;
    time1_buffer[rotor1_data_index] = time1;
    speed_order_buffer[rotor1_data_index]=speed_order;
    rotor1_data_index++;

    update_pwm_flag_rotor1 = 1; // Indicate that rotor 1s speed has been updated
    
}
...
int main(int argc, char **argv) {
    ...
    if (gpioInitialise() < 0) {
        fprintf(stderr, "Pigpio initialization failed\n");
        return 1;
    }
    gpioSetMode(ROTOR1_PIN, PI_INPUT);
    gpioSetMode(ROTOR2_PIN, PI_INPUT);
    gpioSetMode(PWM_PIN, PI_OUTPUT);

    gpioSetISRFunc(ROTOR1_PIN, RISING_EDGE, 0, rotor1_interrupt);
    gpioSetISRFunc(ROTOR2_PIN, RISING_EDGE, 0, rotor2_interrupt);
    ...
    }
```

It fails at the first gpioSetISRFunc

Thanks in advance 

Tom

595NoPower, 2024-08-17:
I want to use GPIO21 as interrupt pin and encounter the exactly same problem

After trying "cat /sys/kernel/debug/gpio", I got

```
gpiochip0: GPIOs 512-565, parent: platform/3f200000.gpio, pinctrl-bcm2835:`
gpio-512 (ID_SDA              )
gpio-513 (ID_SCL              )
gpio-514 (GPIO2               )
gpio-515 (GPIO3               )
...
gpio-533 (GPIO21              )
...
gpio-565 (SD_DATA3_R          )
...
```

and then I tried the following commands
echo 533 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio533/direction
echo 1 > /sys/class/gpio/gpio533/value

GPIO21 actually output high.

So it seems to use GPIO 512-565 instead of 0-54 in my OS which is 
Linux raspberrypi 6.6.31+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.31-1+rpt1 (2024-05-29) armv7l GNU/Linux

but the source code of gpioSetISRFunc uses 0-54 to export, set direction and the edge of GPIO

In this situation, I don't know how to configure interrupt with pigpio.

gpe92, 2024-11-02:
Hi,
I've the same problem with a code which worked fine on the same RPI with previous RPI OS version in 32 bits but no longer with current RPI OS version in 64 bits ...

bennyyyy-x, 2025-02-07:
I was having the same issue and I was able to get `gpioSetISRFunc` to work by shifting the gpio number by 512 in the source code. Inside the `pigpio.c` file and function `intGpioSetISRFunc`, I added the following one line of code.

```C
static int intGpioSetISRFunc(
   unsigned gpio,
   unsigned edge,
   int timeout,
   void *f,
   int user,
   void *userdata)
{
   char buf[64];

   char *edge_str[]={"rising\n", "falling\n", "both\n"};
   int fd;
   int err;

   gpio += 512; // Shifting GPIO numbering to start with 512

   DBG(DBG_INTERNAL,
      "gpio=%d edge=%d timeout=%d function=%08"PRIXPTR" user=%d userdata=%08"PRIXPTR,
      gpio, edge, timeout, (uintptr_t)f, user, (uintptr_t)userdata);
   ...
}
```

Then after installing this version, `gpioSetISRFunc` now works. But I am not an expert, I am not sure if this is a right solution.