joan2937/pigpio #623: Would using GPIO chardev interface ever be considered for pigpiod?

raintonr, 2025-02-02:
Sounds like redundant/cross purposes, right? If so, never mind. But...

I really like the 'Remote GPIO' function of a Raspberry Pi. Aka. `pigpiod`.

I'd like to run this in a lightweight O/S like [OpenWRT](https://openwrt.org/), but that system does not provide `/dev/mem` as default out of the box (they think it's a security concern) so one has to build an image manually to get this which is a bit cumbersome.

What OpenWRT does provide as default is the GPIO chardev interface, which can be accessed with `libgpiod`.

Is there any appetite for making `pigpiod` utilise the GPIO chardev interface (ie. use `libgpiod`) so a package for something like OpenWRT (and other systems where `/dev/mem` is not provided) could be created?

guymcswain, 2025-02-04:
Are you aware of the [lg](http://abyz.me.uk/lg/rgpiod.html) library which is based on libgpio but has API that are familiar to pigpiod.  Remote clients exist for C and Python.  So, are you asking about support for another net client?  Or is there something else about pigpiod that is missing?

raintonr, 2025-02-04:
I hadn't seen lg. Why does this exist alongside pigpiod?

I'm using nodejs so that could be a problem if no package to act as an lg client exists.

I successfully have pigpiod running on OpenWRT with a nodejs client on another host, but... I had to build a custom OpenWRT image to run this as `/dev/mem` isn't enabled as default. This isn't useful when updates come along or for the possibility of sharing pigpio as a package for OpenWRT.

Hence my question is about the possibility of having pigpiod use something other than `/dev/mem`

closed 2025-02-05 by guymcswain