joan2937/pigpio #624: Segfault when setting up multiple callbacks while GPIO events are firing

misterfish, 2025-03-10:
Hi, thanks for the great libary. We have created a NodeJS wrapper for it, which works great, except that we ran into a nasty segfault over the weekend and it appears to come from pigpiod.

Steps to reproduce on a Raspberry Pi Zero 2W:

- start some intensive I2C communication using the hardware chip and without using pigpiod (this will generate lots of events on pins 2 & 3)
- after compiling the C program below, run this loop:

    while true; do if ! ./test-gpio-callbacks; then break; fi; done

On our system this reliably segfaults within about 20 iterations.

We have an ugly workaround but it would be great to know why this is happening.

misterfish, 2025-03-10:
test-gpio-callbacks.c

```
#include <stdbool.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

#include <pigpiod_if2.h>

#define NOISY 0

int edges[3] = {
  RISING_EDGE,
  FALLING_EDGE,
  EITHER_EDGE,
};

static void ac_info (const char *s) {
  fprintf (stdout, s);
  fprintf (stdout, "\n");
}

static void ac_err (const char *s) {
  fprintf (stderr, s);
  fprintf (stderr, "\n");
}

static void ac_die (const char *s) {
  ac_err (s);
  exit (1);
}

static void cb (int pi, unsigned pin, unsigned level, uint32_t tick) {
  if (NOISY) printf (".%d", pin);
}

static int start() {;
  int pi = pigpio_start (NULL, NULL);
  if (pi < 0) {;
    ac_err ("Couldn't open pigpio connection:");
    ac_die (pigpio_error (pi));
  }
  return pi;
}

int main (int argc, char **argv) {
  int pi = start ();
  printf ("started ... ");
  fflush (stdout);
  for (uint8_t gpio = 0; gpio < 28; gpio++) {
    for (uint8_t i = 0; i < 3; i++) {
      int edge = edges[i];
      int rc = callback (pi, (unsigned) gpio, edge, cb);
      if (rc < 0) {
        ac_err ("Couldn't initialize GPIO watcher");
        ac_die (pigpio_error (rc));
      }
    }
  }
  usleep (1e6);
  ac_info ("completed successfully");
  return 0;
} 

```

Compiled as:

`gcc -Wall  test-gpio-callbacks.c -o test-gpio-callbacks  -lpigpiod_if2`

`gcc --version`
gcc (Debian 12.2.0-14) 12.2.0

Using pigpio-1.78 from source, and also 1.79-1+rpt1 from Debian

guymcswain, 2025-03-10:
> Using pigpio-1.78 from source, and also 1.79-1+rpt1 from Debian

How is this reported?   What do you see when you run `pigs pigpv`?

> start some intensive I2C communication at 400kHz using the hardware chip and without using pigpiod (this will generate lots of events on pins 2 & 3)

How are you doing this?  Must it be I2C running or anything that can produce edges on gpio pins?

Does the script run successfully when there is moderate to no gpio activity?

How does this test script expose the problem you are seeing in your real application?   I don't think opening an absurd number of callbacks is realistic.  If you just set up callbacks on gpio 2 and 3 does it work?

Have you tried other hardware versions?  I think I remember that zero 2W silicon is similar to pi3b but I'm not sure.  In any case, I don't have one at hand.

guymcswain, 2025-03-10:
A nitpick perhaps, but callbacks should be cancelled otherwise it could cause a memory leak.   Your script only generates 84 callbacks so probably not relevant to the problem you are seeing.

guymcswain, 2025-03-10:
Although the documentation doesn't mention one callback per gpio, looking at the code it seems there is a `gpioAlert` array of `gpioAlert_t` structures dimensioned to the number of gpio pins.
```C
static gpioAlert_t      gpioAlert  [PI_MAX_USER_GPIO+1];
```

misterfish, 2025-03-11:
> Although the documentation doesn't mention one callback per gpio, looking at the code it seems there is a `gpioAlert` array of `gpioAlert_t` structures dimensioned to the number of gpio pins.
> 
> static gpioAlert_t      gpioAlert  [PI_MAX_USER_GPIO+1];

Interesting catch. We're using the `callback` function from pigpiod_if2.c and not the `gpioAlert` family of functions from pigpio.c though.

Are they related under the hood?

guymcswain, 2025-03-11:
> Are they related under the hood?

Not exactly.  The server accepts requests from a client to report any level changes on one or more gpio.  Then notification messages are pushed to the client.  It is then up to the client as to how to handle these notifications as callbacks depending on the level.  I know that for pigpio-client only a single callback is associated per gpio and the callback itself must handle the edge cases.  The Python client documentation says it can handle more than a single callback per notification.  pigpio.if2.c does not mention this.

So, I think for each exec of `int rc = callback (pi, (unsigned) gpio, edge, cb);`  the server is told to send notifications of a level change for that gpio, which from the server's perspective is redundant per edge case, but then internal to the client, the cb is associated to the particular edge.  That, I assume, can vary depending on how each client handles notifications.

guymcswain, 2025-03-11:
It's also possible the variables (`monitorBits`, etc) controlling notifications are not sufficiently protected by a mutex.

I won't be able to look into it for the next 36-38 hrs.

guymcswain, 2025-03-13:
Are you running 64 bit OS?  If so, can you try your script on 32 bit OS?  (I'm not in a position to run tests at the moment.)

misterfish, 2025-03-13:
> Are you running 64 bit OS? If so, can you try your script on 32 bit OS? (I'm not in a position to run tests at the moment.)

Yes it is 64-bit Raspberry OS. I don't have access to a 32-bit one right at the moment I'm afraid.

guymcswain, 2025-03-13:
I won't be able to run any testing until early next week.  If it runs on 32 bit, my next test is going to be use `stdatomic.h` functions for read/write access to vars monitorBits, notifyBits, etc.

guymcswain, 2025-03-14:
Another thing you can try is to run the pigpio_if2.c script remotely.   This way we can isolate if the crash is associated with the server or the client.

guymcswain, 2025-03-17:
I'm running your scripts on a pi400 but I'm not using I2C traffic for generating gpio callbacks.  I'm creating gpio edges by running multiple waveforms (using DMA) on pigpiod.  Currently I'm running two groups of gpio with 90KHz square waves.  The groups are 180 degrees out of phase.  (The groups are 17, 27, 22 and 23, 24, 25).

But I'm unable to create the crash using this method.  Do you know of any reason why this setup fails to show the problem?  How do you generate the I2C traffic?